xoseperez/espurna #989: Overcurrent protection feature

hyteoo, 2018-06-26:
A useful feature for energy monitoring devices would be a power threshold (in watts) set from the web UI (and of course WS, MQTT, telnet) over which the switch disconnects.

This would be a form of protection for both the powered device and the espurna device (especially for custom hardware where power handling varies substantially depending of relays used)

I'm thinking of 2 parameters here
- power threshold (in W)
- qualification time (in ms)

If ( [power] > [threshold] for more than [qualification time])
{
  switch off
  send MQTT error message
  add to log
}

swilson86, 2018-06-27:
A fuse would normally do that. I would like to see a brownout and over-voltage enhancement if Xose is going to look into your request. It fits into the same category.

Fuses do not protect from brownouts and over-voltage. I have this setup in node-red now and I send an MQTT message to cut the relay in my POW and S31 devices when these events occur (usually during a lightning storm or when power is restored). However, it takes time between the Espurna reports to shut off the relay.

I also watch Watts in node-red and I can tell if my water pressure pump stalls. Node-red shuts off the power if that happens, sends a pump warning MQTT maintenance required message, then it tries to restart the pump 3 times momentarily, and if wattage continues to be high it shuts off the power and sends an MQTT pump failure notice.

hyteoo, 2018-06-27:
NodeRed is also what I'm using now but that is not a safe failsafe
There are many flavours of commercials sockets rated for 10A, 16A, however I wouldn't trust even 10A through most of them.
Therefore I think it would be a good safety boost to have a local monitor that can react quickly in case of overload.

swilson86, 2018-06-27:
I am not sure of the standards in your country on circuit breakers, and am surprised that you do not trust them. In the Americas you can exceed momentarily the Amperage of a circuit. The breaker relies upon heat detection and takes perhaps a second or even less (down to milliseconds) depending on the Amperage to cut the circuit. The design is intentional as motors in particular will create a momentary excess in Amperage up to 300% when starting which will not trip the breaker (as it should be).

The reliance upon heat buildup in a breaker design ensures that a wire (of correct gauge) will not overheat before the breaker trips.

Forgive me for this narrative but I only see complexity in creating digital software to handle things that a fuse or breaker already takes care of. I would suspect that a very large Amperage draw might even cause an Espurna device to malfunction just as it is needed to cut the power. For sure it will cause havoc on the relay inside the device to cut power at the peak moment of Amperage. My faith is with the fuse/breaker to do its' certified job.

hyteoo, 2018-06-27:
@swilson86 I think you misunderstood, I was never talking about circuit breakers
They're kinda the same standard in Europe as they are in the US.

My point was that the devices we use espurna on vary greatly:
- by design: some have 10A rated relay+circuitry, some have 16A, I have some custom boards that can handle only a few amps for lighting purposes, don't need more.
-by quality: some chinese switches have 16A rating on the label but 10A relay and a questionable build quality

The circuit breaker would be a last line of defense in case something really bad happens.
So in order to protect the device itself, this software option would be of great use.