xoseperez/espurna #993: espurna migration

vMeph, 2018-06-26:
ok when i first start using sonoff etc.. i used tasmota and i have migrated to espurna via tasmota web browser with chosing file bin.
now i need it one off my devices to migrate to tasmota again and i did same thing chose bin file from espurna and uploaded the tasmota bin, however it seems that this causes big issues sense it seems that espurna retains some information that will not allow users to migrate

is there any solutions for this?

xoseperez, 2018-06-26:
There is nothing specific in ESPurna to prevent migration to other solutions like Tasmota. What specific problems do you have?

hyteoo, 2018-06-27:
Did you factory reset from the newly flashed Tasmota?
Maybe the EEPROM settings left out from Espurna are interfering with Tasmota

By the way, why did you revert back to Tasmota?

vMeph, 2018-06-27:
@xoseperez  and @hyteoo  the problems i see is the wifi will  show the esp-xxxx only somethimes and if you try log in, will not log in and it will disapear, it seems that some information might be retain on the spiffs, i have tryed the eeprom clear and results is same, very hard to get the oportunity to see the esp on wifi network

one of the reason i  wanted to migrate is cause  after been using espurna for a month i start to see that the sonoffs touchs it seems to freaze alot, i touch them with fingers and will not awake, i have tryed with sleep on 0 and i see same results, i have already migrate some of this toutchs   and with tasmota i dont see no problems so far, but then i would need to wait more time to see, cause i didnt use to have problems with espurna either, only now after awhile i see this happening, 

just not getting why is so hard to revert espurna back to tasmota, cause tasmota to espurna so easy to do it

vMeph, 2018-06-28:
@xoseperez  im having problems with sonoff touch, they freze alot, i go to turn on lights and they dont respond, need to tap alot until it starts to work, dont know where the problem comes from sense it use to work ok, but i can tell that the ones i was able to migrate to tasmota with great dificulty, they work fine and dont freze, not sure why this is happening

xoseperez, 2018-06-28:
Then you are talking about two different things, on one hand the ESPurna image for the Sonoff Touch might not be working ad on the other the migration issue.

Do you have the chance to check the debug log while trying to operate the device?

vMeph, 2018-06-28:
@xoseperez  was able to migrate by changing this parameter in tasmota
`#define CFG_HOLDER             4618              // [Reset 1] Change this value to load SECTION1 configuration parameters to flash`
im still curious about some questions
why did the sonfoffs touch worked ok  with espurna for the first 3 weeks and then start touch frezing with espurna?
why is it working with out freeze now with tasmota?
i realy love espurna, when i first start i use to uses tasmota but then i didnt like that with tasmota using home assistant every time it would do a host reboot the lights would go off, so i moved to espurna and never had that issue, just dont get why is now acting up with buttons not responding when they to long with ou use

xoseperez, 2018-06-29:
You said it was working fine for 3 weeks and then it started freezing? It could be a memory leak. I think I have a Sonoff Touch somewhere in a box. Will give it a try. A debug log will help to find out the problem...

vMeph, 2018-06-29:
yes first 3 weeks was always working great, but then all the 4 touchs i have with espurna started freezing

vMeph, 2018-06-29:
for exemple i have a WC and Office imagen i go to WC tap 4 or 5 times and no light i keep taping and will turn on eventualy,
another thing that i notice is for exemple i tap 3 or 4 times in wc  nothing and then i turn on the office go turn on WC and will turn on right away, same thing happens to office somethimes
but this only happens if they to long with out use, cuase soon i get them to work i can tap and they work perfect, looks like they go into some kind of deep sleep and then they struggle to wake up, i even tryed before with the 0 sleep option to save energy and that didnt happen
i got no idea why has this  problem, i have changed them to tasmota and i got 0 problem i tap 1 time and light turn on, i got no idea why this happen, need to let them now on for a couple time to see if the same thing will happen but so far nothing, all working perfect
i know the all situation seems weird, but is what was happening what i have described no idea
i dont see no one complaining about it so im guessing the problem is on my side, just got no idea why this was happening with the espurna

seetendra, 2018-07-01:
Iam using a Espurna on Sonoff Touch T1 the 3Gang vesion an it working perfectly fine for the last couple of months.
Iam on the latest firmware as well.

xoseperez, 2018-07-01:
I'm not experiencing this issue either. I'm sorry you had a bad experience but I have no way to fix it yet.

closed 2018-08-12 by vMeph