xoseperez/espurna #996: Confirmed working WiON 50051 power strip with the WiON 50055 image

ghvader, 2018-06-27:
I purchased the 50051 power strip. flashed it with the 50055 image and it works.
no energy meter readings.
can you build another image for the 50051 with the energy readings enabled?

xoseperez, 2018-06-27:
I can't find any reference to the power monitor chip it has. Without that info it is hard to add support for that...

ghvader, 2018-06-27:
understood.. my google search indicates it's a HLW8012
if you need me to take it back apart to verify I will.

xoseperez, 2018-06-28:
Do you have the reference to where it says it's a HLW8012? If thats true we would need to know what GPIOs are connected to the SEL, CF and CF1 pins in the HLW8012...

ghvader, 2018-06-28:
4 XPD_DCDC / GPIO16 PF HLW8012 CF power
5 MTMS / GPIO14 VS GPIO05 HLW8012 Sel output
6 MTD1 / GPIO12 VC HLW8012 CF1 voltage / current

https://abzman2k.wordpress.com/2017/03/16/eco-plug-esp8266-hack-finally/

xoseperez, 2018-06-28:
If that's true then test this configuration. I'm still lacking calibration values but you should be able to calibrate the unit via the web interface...

```
#elif defined(WION_50055)

    // Info
    #define MANUFACTURER        "WION"
    #define DEVICE              "50055"

    // Buttons
    #define BUTTON1_PIN         13
    #define BUTTON1_RELAY       1
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    // Relays
    #define RELAY1_PIN          15
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1

    // HLW8012
    #ifndef HLW8012_SUPPORT
    #define HLW8012_SUPPORT     1
    #endif
    #define HLW8012_SEL_PIN     14
    #define HLW8012_CF1_PIN     12
    #define HLW8012_CF_PIN      16
```

xoseperez, 2018-07-01:
Anyway, re-reading the post it looks like there is another microcontroller between the HLW8012 and three ESP8266 so protocol might not be the same... @ghvader can you please confirm if the configuration in my previous post seems to work (remember you might have to calibrate the values, then try with different known loads).

See #48.

ghvader, 2018-07-02:
@xoseperez I have tried twice to roll my own following the arduino method. but it seems to brick each time i flash it. (never boots up) the compile goes okay with no errors and the flash seems to go okay with no errors. but it never boots. if i flash the default pre-built wion binary you made, it works fine. i have not had a chance to try a third time as of yet.

BenchPressesBooks, 2023-08-16:
I have not been able to determine how the microcontroller works yet that interfaces with the temp sensor. Has anyone had luck with this? These devices are all currently DIRT cheap and very nice.