xoseperez/espurna #1000: Sonoff-RF RF not supported?

Geroldswil, 2018-06-28:
Hi all,
I just flashed espurna-1.13.0-itead-sonoff-rf.bin to a brand new Sonoff RF.
Status shows the correct FW name and version.
But I miss the RF menu point, as it exists on the RF bridge.
Is RF not supported with this FW or am I missing something?
Thanks in advance
Ronny

xoseperez, 2018-06-28:
There is no RF menu in the Sonoff RF image because the RF feature is completely handled by the secondary microcontroller. The ESP8266 sees that controller as a button, so pairing and remote functions should work the same as with the stock firmware...

closed 2018-06-28 by xoseperez

Geroldswil, 2018-06-30:
I confirm that the pairing is working and that the red led is quickly flashing during reception of the code.
But the relay does not switch. Manually (by switch) is working fine.