xoseperez/espurna #1001: Thermistor attached to the ADC (analog pin)

nqdanyb, 2018-06-28:
I have a heat resistor(Thermistor) attached to the ADC, How to display the temperature with the unit
  Celsius on web interface and publish via mqtt to home-assistant? (sorry for my english) Thanks for all the help

xoseperez, 2018-06-28:
@nqdanyb if you enable ANALOG_SUPPORT in sensors.h you will get the raw ADC value. The temperature will depend on the thermistor you are using.

nqdanyb, 2018-06-28:
@xoseperez Thank you for taking care of my problem. I have enable ANALOG_SUPPORT in sensors.h and i have get the raw ADC value. but I do not know how to display it in degrees C format on the web. In order to process these raw data, we need to use formulas to convert to c. But I do not know where to put these recipes. (I'm a newbie).Here is my code used with arduino:

// Pin that the thermistor is connected to
#define PINOTERMISTOR A3
// Nominal temperature value for the thermistor
#define TERMISTORNOMINAL 10000
// Nominl temperature depicted on the datasheet
#define TEMPERATURENOMINAL 25
// Number of samples 
#define NUMAMOSTRAS 5
// Beta value for our thermistor
#define BCOEFFICIENT 3977//3977
// Value of the series resistor
#define SERIESRESISTOR 10550 //Thay đổi con số này, càng giảm nhiệt độ càng tăng

int amostra[NUMAMOSTRAS];
int i;
void setup(void) {
Serial.begin(9600);
//analogReference(EXTERNAL);
}

void loop(void) {
float media;

for (i=0; i< NUMAMOSTRAS; i++) {
amostra[i] = analogRead(PINOTERMISTOR);
delay(10);
}

media = 0;
for (i=0; i< NUMAMOSTRAS; i++) {
media += amostra[i];
}
media /= NUMAMOSTRAS;
// Convert the thermal stress value to resistance
media = 1023 / media - 1;
media = SERIESRESISTOR / media;

//Calculate temperature using the Beta Factor equation
float temperatura;
temperatura = media / TERMISTORNOMINAL;     // (R/Ro)
temperatura = log(temperatura); // ln(R/Ro)
temperatura /= BCOEFFICIENT;                   // 1/B * ln(R/Ro)
temperatura += 1.0 / (TEMPERATURENOMINAL + 273.15); // + (1/To)
temperatura = 1.0 / temperatura;                 // Invert the value
temperatura -= 273.15;                         // Convert it to Celsius

//Serial.print("The sensor temperature is: ");
Serial.println(temperatura);
//Serial.println(" *C");

delay(10);
}

poulch74, 2018-06-28:
see in AnalogSensor.h function named "value".  or you can create your own sensor...

xoseperez, 2018-06-29:
Latest version in `dev` branch supports NTC sensors (thermistors), please check the NTC section in the `sensors.h` file.

nqdanyb, 2018-07-02:
@xoseperez. Thank you very much. I upgraded to the dev branch version and had the NTC sensor. I will test it for a while. Ask me a little more. Does ESPurna support I2c connectivity with arduino? (arduino nano, pro mini ...). If there is support for connecting to arduino it is great. The number of analog sensors and the number of control pins will be expanded.

nqdanyb, 2018-07-03:
Here is my ESPurna. NTC works fine but NTP does not:
[image]

Here are my NTP settings When compiled with the NtpClient-develop library:
[image]

But when compiling with the NtpClient-master library as instructed in the platformio.ini file, there was an error: 
Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), 2, v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

Build options changed, rebuilding all
C:\Users\Dell\Downloads\espurna-dev\code\espurna\ntp.ino: In function 'void _ntpStart()':

ntp:48: error: 'class NTPClient' has no member named 'setNTPTimeout'

     NTP.setNTPTimeout(NTP_TIMEOUT);

         ^

C:\Users\Dell\Downloads\espurna-dev\code\espurna\ntp.ino: In function 'void _ntpConfigure()':

ntp:79: error: 'class NTPClient' has no member named 'setDSTZone'

     NTP.setDSTZone(dst_region);

         ^

Multiple libraries were found for "JustWifi.h"
 Used: C:\arduino-1.8.5\libraries\justwifi-master
 Not used: C:\arduino-1.8.5\libraries\JustWifi_2.0.0_1282
exit status 1
'class NTPClient' has no member named 'setNTPTimeout'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

How can I synchronize time with NTP?

xoseperez, 2018-07-03:
Default NTP server (`pool.ntp.oro`) should work in most cases since its an alias for a server pool. The build error is because you might be using the master branch from the original repo but ESPurna uses a forked one and a specific revision (https://github.com/xoseperez/NtpClient.git#0016a59) .

closed 2018-07-10 by xoseperez