xoseperez/espurna #998: TYWE3S-based Estink WiFi Power Strip

tty74it, 2018-06-28:
Hi, i've tried to use ota use the firmware from release "espurna-1.13.0-estink-wifi-power-strip.bin"
but when i try to install via ota or via web an error occour (error 9 o error 4 in telnet session)

I've tryed to compile my self and the same error occour.
I've tryed to install a similar power strip (  espurna-1.13.0-zhilde-eu44-w.bin ) and all is ok
but the gpio is different and some relay works and the other no

I've also tryed to start from espurna-core (minimal space) via ota but the result is the same

Thank's
tty74it

xoseperez, 2018-06-28:
Error 4 means UPDATE_ERROR_SPACE which means there is not enough room in the flash to do OTA. You must do a two steps update to be able to do OTA in this case: https://github.com/xoseperez/espurna/wiki/TwoStepUpdates

Error 9 means UPDATE_ERROR_NEW_FLASH_CONFIG, that could mean the memory layout is not the same... There could be a definition problem here since I see in the platformio.ini file that the board is defined to have 1Mb chip but the memory layout is a 4M1M... I'll check this

xoseperez, 2018-06-28:
Just for reference:

https://github.com/esp8266/Arduino/blob/0f34bd85d8ef05ef436538190d86e674a8039a0f/cores/esp8266/Updater.h#L8-L19

xoseperez, 2018-06-28:
OK, I have found a problem with the 1.13.0 version for the ESTINK and the NEO COOLCAM devices. Apparently, the image reports a wrong filesize with only about 283Kb of free room for OTA whilst current core version is 296Kb... 

The solution would we to either flash it wired with the latest version in dev or do a two-step update like in my previous post but with an extra lightweight image I'm linking here that's only 277Kb (no telnet connection thou). 

http://tinkerman.cat/espurna-1.13.1a-espurna-core-1MB-extra-light.bin

tty74it, 2018-06-28:
Yes i confirm, now all is ok.
I've used extra-light and -ota from myself compile

Thank's
tty74it

closed 2018-06-28 by xoseperez