xoseperez/espurna #999: LINGAN SWA1 broken

cajadelapices, 2018-06-28:
Hi,

I have tested the last version 1.1.13 with LINGAN SWA1 and after a factory reset, the board does not work anymore. I have checked and the ESP is burnt!

Any idea?

Thanks!

xoseperez, 2018-06-28:
What do you mean by "burnt"? It does not even let you flash it wired?

xoseperez, 2018-06-29:
No, a factory-reset only deletes the EEPROM contents... nothing else. I can't see how this could burn the chip...

ColinShorts, 2018-06-29:
How did you connect to the device, did you use 5V serial/power when it should only be 3.3V? Is it possible that the device got shorted out by resting on something conductive?

Did you still have the serial connected when it was attached to mains voltage?

jsponz, 2018-06-29:
I used 5V for flashing:

https://nathan.chantrell.net/20170528/lingan-swa1-plug-in-wi-fi-socket-with-mqtt-esp8266/

After flashing, I put everything inside the housing and I closed it. No serial connection left. In fact, I did not solder any wire or connector to the board. 

I connected it to 230AC and it worked. From the website and using the button, I could switch on and switch off the relay (and the load). As soon as I made a factory reset, it did not work anymore. So I guess the trigger for the "burnt" was the factory reset. 

Could it be a faulty batch?

closed 2018-07-10 by xoseperez