xoseperez/espurna #1003: Alexa per Sonoff ch4pro_dual R2_Rf bridge

Carmelo66, 2018-06-29:
Hi, I'm a beta tester of Alexa and with Sonoff basic with Espurna everything works ok and it's perfect, but with the sonch 4ch pro + Sonoff dua R2 and Rf Bridge the command is processed and confirmed but the relays are not activated.
How can I fix this?

xoseperez, 2018-06-29:
Can you check the debug log to see if the commands are being received by the device? Does Alexa respond with an "ok"?

Carmelo66, 2018-06-29:
The command is received and it is OK, but the relay always remains Off.
I tried to give the command to Alexa while the relay is access and she gives ok and always turns the relay off

Carmelo66, 2018-06-29:
[953718] [ALEXA] Device #1 state: OFF

Carmelo66, 2018-06-29:
Is Sonoff dual R2

xoseperez, 2018-06-29:
Just checked with the RFBRidge and it's working fine for me... What Amazon device do you have? Current implementation might not be compatible with v2 devices.

xoseperez, 2018-06-29:
Please, add some more line from the debug log, what happens before? and after?

Carmelo66, 2018-06-30:
I have 6 devices Sonoff basic version 1, 2 devices Sonoff ch4pro version 1, 4 devices Sonoff Dual R2 version 1 and 1 device Sonoff Rf bridge version 2, all with Espurna 1.13.0 and all devices are detected by Alexa app with Echo device but only the  Sonoffs basic connected to the lights work.

Carmelo66, 2018-06-30:
For Sonoff Basic the log is this:
[515739] [ALEXA] Device #0 state: ON
[515742] [RELAY] #0 scheduled ON in 0 ms
[515754] [RELAY] #0 set to ON
[515757] [MQTT] Sending Espurna_6/relay/0 => 1 (PID 1)
[515759] [MQTT] Sending domoticz/in => {"idx": 35, "nvalue": 1, "svalue": ""} (PID 1)
[515925] [MQTT] Received domoticz/out => {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014073",
   "idx" : 35,
   "name" : "Luce_Salotto",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}

Carmelo66, 2018-07-11:
All devices that have only one relay (type Sonoff Basic) with Alexa work perfectly, but devices that have 2 or more Alexa relays do not work, only reads the status of the relay if it is ON or OFF.

closed 2018-09-16 by stale[bot]