xoseperez/espurna #1004: duplicate switch?

lanceji, 2018-06-29:
i close the discover for espurna console and add the config to ha configuration.yaml.
but the homeassistant states ui has the duplicate switch, what is up?

one can control,
another can not control, it is can view states.

how can i del the duplicate switch.

lanceji, 2018-06-29:
switch:
  - name: EastBedroom-TV
    platform: mqtt
    state_topic: Home/EastBedroom-TV/relay/0
    command_topic: Home/EastBedroom-TV/relay/0/set
    payload_on: 1
    payload_off: 0
    availability_topic: Home/EastBedroom-TV/status
    payload_available: 1
    payload_not_available: 0

lanceji, 2018-06-29:
the config is auto generate

lanceji, 2018-06-29:
the entity_registry.yaml file auto generate everytime

switch.eastbedroomtv_2:
  config_entry_id:
  name:
  platform: wemo
  unique_id: 221703K044EF75
switch.eastbedroompow_2:
  config_entry_id:
  name:
  platform: wemo
  unique_id: 221703K04A8F29

lanceji, 2018-06-29:
i used espurna-1.13.0-itead-sonoff-basic.bin.

xoseperez, 2018-06-29:
Could it be that you have Alexa enabled in your ESPurna device and HA is detecting it and adding it as a switch?

lanceji, 2018-06-30:
yes，i am open it,  thanks
i try close it later..

lanceji, 2018-06-30:
thank you very much.....it is right...great job...

closed 2018-07-01 by xoseperez