xoseperez/espurna #1010: Generate password on first-time-use page

mcspr, 2018-06-30:
From https://github.com/xoseperez/espurna/commit/ad623524f515ee6157eb3277d54a0a83e357a7cb#commitcomment-29176290

Some passwords, like UUID, do not conform current password policy.
Some users do not use password managers.
Some browsers do not provide facility to auto-create password (try right clicking password field in Chrome)

As proposed in commit comment, what about generating password instead? For example / [Live demo](https://jsfiddle.net/nm8bc7ue/1/)
```js
// !\"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\\]^_`abcdefghijklmnopqrstuvwxyz
var chars = new Array(94).fill(0).map((_, i) => String.fromCharCode(i+33))
var source = new Uint32Array(10);
var pass = new Array(10);

window.crypto.getRandomValues(source).forEach(function(value, i) {
    pass[i] = chars[value % 94];
    console.log(i, value % 94);
});
console.log(pass.join(""));
```

mcspr, 2018-08-31:
ping? @xoseperez

xoseperez, 2018-08-31:
I do not think this is necessary or even good. My experience says that this will lead to tens of messages saying "I've been locked out of my device". Unless it's optional (like a button saying "auto-generate password") I prefer changing the regexp to support UUID.

mcspr, 2018-09-02:
Ok, maybe this makes more sense on some 'configurator' of sorts to initially setup multiple devices, which gathers common configuration parameters like mqtt / ntp / wifi settings etc and applies them + generated password. At the very least wifi password can be extracted if such 'configurator' saves initial config (or for example immediately display qr code that will be saved by the phone) to avoid credentials loss.

*sidenote about passwords in general:* UUID is just uppercase, numbers and '-', which do not match requirement to have lowercase letters. Should policy restrict anything but wifi passphrase requirements? i.e. printable ascii characters is much shorter regexp to maintain: `/^[\x21-\x7e]{8,63}$/;` (TIL regexp can contain raw codes) 
It's not that it is complicated, but current policy still allows something like 'Admin123' that is not much far from 'p@ssw0rD' / 'password' that's just switches app specific default password for common default password. Long passphrases like 'gusto grounder ovule postal reality overdone' too require tweaking to arbitrary insert number / special character

xoseperez, 2018-09-04:
Again, it's OK to enforce a strong password policy. But last time, about 6 months ago we softened it... Anyway, the problem is not the password strength but the fact that the default build does not support SSL, so there is little point in talking about passwords...

mcspr, 2018-09-07:
Hence my point about enforcing it at all. Regexp can only notify about possible repercussions but otherwise do nothing more than force technical length limit for manual and fallback AP mode sake.

BTW now that Chrome 69 is out, autocomplete='new-password' triggers builtin generator via pop-up alert under the input box. Only on PC now but i guess it is in the works for mobile versions, as the is optional flag to turn this functionality on.
But i got the impression that form inputs and HTTP auth are separate entities - saving form data (with additional `<input name='user' value='admin'/>` before) and triggering password generator does not autocomplete in HTTP auth popup, while chrome://settings/passwords does list saved admin/password pair.

Manual generation looking good otherwise (with additional feature of 'reveal password' button near password input), only problem being how to handle browser's generator.

mcspr, 2018-09-08:
Gist of changes: [dev...mcspr:webui-generate-password](https://github.com/xoseperez/espurna/compare/dev...mcspr:webui-generate-password)
I'll try to find if it is possible to save http credentials somehow. Otherwise, reverting to autocomplete='false' to disable popup on input field

mcspr, 2018-09-15:
No luck in saving http through form 😞 Maybe it can be done, I did not find it. So the gist is:
- password reveal eye-button. (funnily enough, edge on windows already has this feature, but css can disable it to not overlap)
- first-time screen centered, additional button for generation, reveals text when pressed
- minor placeholder changes to distinguish revealed text and placeholder='...'

Should it be separate or as one pr?

xoseperez, 2018-09-16:
Go ahead and do a single PR.

PR #1203 (2018-09-20): Update password UI
Resolves #1010 

* revise first-time-use screen
* generate password (policy applies)
* hide/reveal password settings
* changed autocomplete='...' for some inputs to avoid accidental autocompletion
* split '#saveForm' into multiple '.form-settings' forms to avoid accidental autocompletion across panels

It would be nice if someone can test this on Safari 🍎
closed 2018-09-20 by xoseperez