xoseperez/espurna #1011: WiFi pass0 lost after reboot?

Zebble, 2018-07-01:
I'm using a Sonoff Touch clone.  After reboot, the pass0 value changes to "pass0": "9486".  Any ideas why?

This is what I'm using.

https://www.amazon.ca/gp/product/B07C9DZFHV/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1

Seemed to be fine with Tasmota...

Using the latest release of Espurna (1.13.0)

Thanks!

xoseperez, 2018-07-01:
Your browser is probably populating the password field with a saved password.

Zebble, 2018-07-02:
This seems to be exactly the problem.  I loaded up another browser (FireFox) and works as expected.  Not sure how I saved that key, as it's not a number I've ever used.  But, looks like human error.

Thanks for the help!  Closing.

closed 2018-07-02 by Zebble