xoseperez/espurna #1013: Alexa support keeps crashing a generic ESP12e

bajo, 2018-07-01:
I have a generic ESP12e with a self-compiled espurna (87859c5ad89e2544cb2d65882f25185bc167528a) that keeps restarting after I connect it to my WiFi.

The output I get on the serial monitor is:
```
Exception (28):
epc1=0x40220c23 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000c depc=0x00000000

ctx: cont 
sp: 3fff1a70 end: 3fff1ca0 offset: 01a0

>>>stack>>>
3fff1c10:  39313a30 0a0d3030 6e6e6f43 69746365  
3fff1c20:  203a6e6f 3fff082c 3fff0810 40220d0d  
3fff1c30:  3ffea320 0301a8c0 3ffea320 0301a8c0  
3fff1c40:  00009a2c 3fff1c30 3fff0a70 4020410c  
3fff1c50:  3fffdad0 3fff08a8 00001aac 3fff0c6c  
3fff1c60:  3fffdad0 3fff08a8 0000000b 4020bb38  
3fff1c70:  3fffdad0 3fff08a8 0000000b 402038e4  
3fff1c80:  3fffdad0 00000000 3fff0c66 402270a4  
3fff1c90:  feefeffe feefeffe 3fff0c80 401006fc  
<<<stack<<<

```
EspStackTraceDecoder shows:
```
Exception Cause: 28  [LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads]

0x40220ce3: fauxmoESP::_nextUDPResponse() at /home/bajo/projects/doris/switch/espurna/code/.piolibdeps/FauxmoESP/src/fauxmoESP.cpp:388
0x40220dcd: fauxmoESP::handle() at /home/bajo/projects/doris/switch/espurna/code/.piolibdeps/FauxmoESP/src/fauxmoESP.cpp:388
0x4020410c: ntpSynced() at /home/bajo/projects/doris/switch/espurna/code/espurna/settings.ino:423
0x4020bc04: alexaLoop() at /home/bajo/projects/doris/switch/espurna/code/espurna/settings.ino:423
0x402038e4: loop at /home/bajo/projects/doris/switch/espurna/code/espurna/settings.ino:423
0x40227164: loop_wrapper at /home/bajo/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:123
0x401006fc: cont_norm at cont.S.o:?
```

Some people mentioned that using core 2.4.1 is more stable and could solve this issue I ugraded from 2.3.0 to 2.4.1, however the problem remained. The solution that worked for me was to remove Alexa support with the following build options. ``-DALEXA_SUPPORT=0 -DALEXA_ENABLED=0``

svwhisper, 2018-07-28:
Thanks for the workaround.  Removing Alexa support did the trick for me too.

xoseperez, 2018-08-29:
Latest code in `dev` branch uses fauxmoESP version 3.0.X which changes completely... Can you check against new version?

bajo, 2018-08-29:
Might not be until Sunday that I have time and access to one of my ESPs. But I will check and report back the result.

mcspr, 2018-11-28:
@bajo Any update on this? Can you try latest `dev`? Or just this change -  https://github.com/xoseperez/espurna/commit/56a8010a4def02f40a5979ecb879534d80e466d8