xoseperez/espurna #1014: Make rfbridge module generic RF<->MQTT

mcspr, 2018-07-01:
Current RF_SUPPORT (rf.ino) does not provide MQTT API nor does allow sending data. And current ITEAD_SONOFF_RFBRIDGE (rfbridge.ino) allows to use rcswitch directly. They can be combined into one module.

Some things to change:
- decide at compile time what logic to use (rcswitch or efm)
- rename either RFB_DIRECT or RF_SUPPORT
- combine RF_PINs and RFB_TX_PINs definitions (pending gpio settings in config, namespace is the same)

InduPrakash, 2019-04-11:
I don't think there is any rf.ino module, right ? I too was quite confused about what RF_SUPPORT is meant for. I wanted to use basic 315Mhz Tx/Rx modules to control a switch so ended up faking it as ITEAD_SONOFF_RFBRIDGE with RFB_DIRECT and everything seems to be working even the MQTT part.

mcspr, 2019-04-12:
It was combined with rfbridge: https://github.com/xoseperez/espurna/pull/1435
Although it is still kinda confusing. See #1693, some tests would be appreciated.

closed 2019-11-09 by mcspr