xoseperez/espurna #1016: ntp

Slowsteve, 2018-07-02:
Since updating to 1.13 I get "[NTP] Error: NTP server not reachable". I've tried different servers and yet the time remains at 2018-06-30 10:13:51. Don't think I've done anything to mess it up but am pretty new to this so maybe I unknowingly have. Any thoughts please?

xoseperez, 2018-07-02:
I've also seen the message but normally it ends up connecting after a few retries.

AlbertWeterings, 2018-07-02:
I have been seeing issues connecting to NTP and other stability problems before as for example mentioned in https://github.com/xoseperez/espurna/issues/842 .

What I found  is that when in the ARP table of my router a MAC address is linked to a IP and a ESPURNA device with a different MAC address is using the same IP it will not connect to NTP to get the time. In my situation the router is build on ClearOS 6.9. I noticed that when I manually delete the MAC and IP address from the ARP table NTP starts working again. I don't think the problem in my case is with ESPURNA it has something to do with the router not flushing the ARP table. 

So in your case I would advise to give your ESPURNA device a IP that is not used before in your network if at that point NTP starts working find a way to flush the ARP table of your router.

closed 2018-09-07 by stale[bot]