xoseperez/espurna #1017: OTA Authenticating...FAIL

pcornill, 2018-07-02:
This is the 4th time it happens to me so I believe something is wrong. Did anyone of you experienced this issue?
It happens to me with Sonoff Basic, TH and POW.
It is when updating from 1.14 to 1.16 but that I am not 100% sure it was the case for previous OTA errors.
_Authenticating...FAIL
14:15:02 [ERROR]: No Answer to our Authentication_
==> this is not a password issue because when I put the wrong password, I got this:
_Authenticating...FAIL
14:22:41 [ERROR]: Authentication Failed_

xoseperez, 2018-07-03:
Does it always happen or it is random? It could be the device not responding on time... but then you should be able to retry and succeed.

pcornill, 2018-07-03:
It is not random. You can believe me I tried everything I could to avoid opening the box. I rebooted several times but always got that error. Same error from Telnet, nothing additional.

xoseperez, 2018-07-03:
Does the device have enough free space for OTA?

One thing: those version numbers (1.14, 1.16) are not for ESPurna...

pcornill, 2018-07-03:
Yes Espurna, I corrected the version (my mistake).
In term of free memory, I believe they do. I am using the firmware you developed and nothing else.
If it works with any Sonoff Basic, it shouldn't be an issue, right?

xoseperez, 2018-07-03:
Can you log into the device and check the debug log (via telnet or in the web interface) and see what happens when you try to flash it OTA?

pcornill, 2018-07-05:
As I said earlier, I also try to log with Telnet and didn't get any additional message than the one Arduino IDE shows me. I do not have ways to reproduce because I uploaded the latest Espurna using serial port - which was successful -
You may want to close that issue but at least we have a record and someone may see the same problem later on. Thx Xose for your prompt support.

reddevil156, 2018-07-08:
I got the same message today when trying to do a OTA with the arduino IDE. I used the dev brach 1.13.1 for a magic_controller_20
After the compilation was completed a window with the password prompt appeared. I put in my password and got the error:
Authenticating...FAIL
22:45:07 [ERROR]: No Answer to our Authentication

The log file in espura shows the following message:

[115772] [WEBSOCKET] #5 connected, ip: 192.168.160.43, url: /ws
[156618] [OTA] Error #1: Begin Failed
[156626] [EEPROM] Reenabling EEPROM rotation

I think there is too few space for the flash but shows the wrong error message.

xoseperez, 2018-07-10:
I'm keeping it open. There are a number of reasons for the error #1:

* An update is already running
* The boot mode is not right to reboot
* The new image size if 0 
* Flash configuration is wrong (larger than the physical size)
* New image size is larger than the free space

Most of the time it would be the latest reason.

closed 2018-09-15 by stale[bot]