xoseperez/espurna #1018: GPIO extender support

evolucja, 2018-07-02:
Hello,
does ESPurna support any GPIO extender like this:
https://pl.aliexpress.com/item/PCF8574-IO-Expansion-Board-I-O-Expander-I2C-Bus-Evaluation-Development-Module/32441163158.html

?

I would like to use about 14 devices with one Wemos D1 and therefore I need more GPIOs...

Thanks

Misiu, 2018-07-03:
Maybe MCP23017 could be supported? https://www.aliexpress.com/item/MCP23017-Serial-Interface-Module-IIC-I2C-SPI-MCP23S17-Bidirectional-16-Bit-I-O-Expander-Pins-10Mhz/32866459372.html
Adafruint has a library that supports it and examples look simple:
https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library/blob/master/examples/button/button.ino

xoseperez, 2018-07-03:
See #199. But prior to support it a higher level of abstraction must be added to the firmware to allow using "virtual" GPIOs

closed 2018-07-03 by xoseperez

MaliMrav, 2019-02-12:
Firstly, let me say "I've evaluated all flavours: ESPEasy, Tasmota and Espurna and I must say, Espurna is rock solid!"
It integrates so well with Domoticz, Homeassistant and the MQTT actually works in receiving AND sending to home automation systems (unlike ESPEasy). So now I'm 95% there. The thing I'm missing is virtual GPIOs in Espurna. 
This is the only thing it needs to be leaps and bounds ahead of the others.
Please, please, please do the abstraction to include Virtual GPOIs and add support for PCF8574 (ESPEasy has that support but their MQTT can only write to HA systems. You cannot control actuators on ESPEasy via MQTT.