xoseperez/espurna #1020: Switching RGB channels.

T-bond, 2018-07-03:
Hello.

I had successfully built and uploaded this fantastic firmware to my RGB led controller.

My problem is (it was with the controller's default software too), that the Blue and Green channels are swapped. When I set it to Green it is blue and when I set it to blue it is green. (Red is in the correct channel)

My question is: is there an easy way to swap them in the firmware, so it will light in the correct colors? (with rebuilding it)

Misiu, 2018-07-03:
@T-bond can't You fix connector (just swap G and B pins)? It is the easiest fix.

T-bond, 2018-07-03:
I would like to swap in the software if it is possible, but if it not, then I will have to.
I don't want the physical swap because I have a lot of from this controller, and a software swap would be way easier. [The controller](https://photos.app.goo.gl/5uK6RqbJL5ohyU3Q9)

T-bond, 2018-07-03:
Oh. I just find: https://github.com/xoseperez/espurna/issues/680

closed 2018-07-03 by T-bond

Misiu, 2018-07-03:
@T-bond didn't know that's possible. Thanks for the link :)