xoseperez/espurna #1021: Sonoff RF

Geroldswil, 2018-07-03:
Hello!
I'd like to come back on my issue 1000.
I have verified on two samples (Sonoff RF with 1.13.0) that learning the RF code is working.
Receiving the code is indicated by quickly flashing of the red LED, but the relay is not operated.
Manually switching (by button) and via WEB-IF is working fine.
???
Thanks!

Geroldswil, 2019-12-31:
...any chance this will be fixed once?
Thanks & Happy New Year

mcspr, 2019-12-31:
Not sure what to fix exactly. Sonoff RF, same as T1, emulates button press on the device when RF code is received:
```
[334519] [BUTTON] Button #0 event 1
[335020] [BUTTON] Button #0 event 2
[335022] [RELAY] #0 scheduled ON in 0 ms
[335034] [RELAY] #0 set to ON
[336037] [RELAY] Setting relay mask: 0b1
[337001] [BUTTON] Button #0 event 1
[337503] [BUTTON] Button #0 event 2
[337505] [RELAY] #0 scheduled OFF in 0 ms
[337516] [RELAY] #0 set to OFF
```

mcspr, 2020-01-03:
Perhaps button mode needs pull_up mode set?
https://gitter.im/tinkerman-cat/espurna?at=5df8a3a20dc628523e4de80c
https://github.com/xoseperez/espurna/blob/62ad7da332f3f904ad8241a2b738be9820b196e4/code/espurna/config/hardware.h#L336
(adding `BUTTON_SET_PULLUP`)

Geroldswil, 2020-01-04:
Just tried again with 1.14.1. Same as before. According to the schematic, there is a (HW)pull-up on S1 as well as on E-FW to (pin 15). The problem might be in the EFM8BB1? Is it possible to modify it's code?

mcspr, 2020-01-29:
https://www.itead.cc/wiki/Sonoff_RF also mentions clearing procedure. Maybe code is learned incorrectly?
Also see #2125. I am not really sure the issue is with the band and not that with non-fixed code remote.

I don't think EFM8 reprogramming for SonoffRF came up so far, but it is theoretically possible:
https://www.itead.cc/wiki/images/f/f8/Sonoff-Schmatic.pdf
Assuming TP3 & TP4 are properly linked to the programming interface.
https://github.com/Portisch/RF-Bridge-EFM8BB1 is a project for RFBridge board with uses a similar chip. It does not deal with internal memory (to store codes), so idk how that works. There is no UART handling required, so some part of the firmware can be simplified.