xoseperez/espurna #1024: Add support for Sonoff iFan

xoseperez, 2018-07-03:
Summer is here.

ColinShorts, 2018-07-03:
Shipping to the UK seems a bit steep 😥
The iFan02 looks interesting though 🤔

thebradleysanders, 2018-07-03:
Its seems Tasmota recently added the Ifan02 to their supported devices list, it looks like the device has an esp8285. Interested in seeing it added to Epurna as I see it better than Tasmota.

xoseperez, 2018-07-03:
I have just added support for the IFAN02 to the `dev` branch. Web interface is not yet ready but I'd like to know if it works. Can anyone test?

1043717432, 2018-10-05:
@xoseperez  Testing, the web page operation button is perfect, but it can't feedback with the ifan02 remote control.
Tasmota supports RF remote control feedback for ifan02, and when remote control is performed, there is a corresponding action response in the web page.

1043717432, 2018-10-05:
@xoseperez  Tasmota supports feedback from the ifan02 radio frequency remote control, and when the radio frequency remote control operates, there is a corresponding action response in the web page.

arielo72, 2018-11-12:
Hello everyone, I´m a newbie in electronics, do you know if it's possible control the light switch with a push button_ so I can also control the light replacing the actual light switch by a push button?
thanks a lot in advance!

thebradleysanders, 2018-11-13:
@arielo72, is this question related to the Sonoff Ifan for Espurna in general?

arielo72, 2018-11-15:
Hello @thebradleysanders yes, would be nice to have it. As the controller also control the light, would be nice to control the light from the wall switch as well.
thnaks

reaper7, 2019-02-23:
@xoseperez - does the current code for IFAN2 also accept commands from the remote control?
Watching hardware.h looks like he should... can you confirm?

fquinto, 2019-03-12:
> I have just added support for the IFAN02 to the `dev` branch. Web interface is not yet ready but I'd like to know if it works. Can anyone test?

Hi, I like to inform all that I have one Sonoff iFan02. I tested using Tasmota and it's running well (with the remote controller included). But I prefer to use Espurna, then I sorry to confirm that this development need changes. I will try to develop this changes in order to provide a running Sonoff iFan02 in Espurna.

fquinto, 2019-03-17:
Again "summer is near", and the good weather needs to improve the code inside **Sonoff iFan02**.
I do and small improvement here: https://github.com/xoseperez/espurna/pull/1639

**Problems found when we use remote controller**
Problems come when you need to sync the small remote controller. After a long time trying for me the procedure has been:

- Using the original firmware (from Itead Sonoff).
- Check battery: not less 2.85v and not 3.3v. Only runs very near 3v.
- Put the new battery: press the hide button (open remote controller for that)
- The LED in remote controller is like "searching"
- Connect the ifan02 main unit to the AC current or if connected, reboot. (WARNING: be careful with this AC!)

I think the join was during the reboot of the main unit. Hope this helps.

But, the remote control, part of them is controlled by the MCU and this go outside of the ESP.
Sometimes the remote controller stops running and we need to reboot the main units.

When we press "key 3" in remote controller I can see in the debug that quickly sends the button 1 and button 3, this is normal. But I think something occurs inside EEPROM? And the main firmware still running well and the remote controller stops. If you reboot, you get again  a running remote controller.

For the moment I don't have more time to debug this issue and form me it's better to use the MQTT function to control the fan speed and light.

Here exists and schematics of Sonoff iFan02 in PDF format that helps me: https://github.com/arendst/arendst.github.io/blob/master/media/ifan02/iFan02Schematics.pdf

Enjoy!!

peterhoeg, 2019-07-01:
Has anybody tried with the new [ifan03](https://www.itead.cc/sonoff-ifan03-wifi-ceiling-fan-light-controller.html)?

UmerFraz, 2019-08-07:
my iFan03 arrived just yesterday to UK (bought from Banggood). Now to wait for the support :) which hopefully comes before summer ends here. I saw on tasmota's forum that iFan02s template wont work with 3.

Hedda, 2020-06-26:
Any news on support for Sonoff iFan03 by ITeam in ESPurna?

https://www.itead.cc/sonoff-ifan03-wifi-ceiling-fan-light-controller.html 

It is supported in Tasmota

 https://tasmota.github.io/docs/devices/Sonoff-iFan03/

mcspr, 2020-06-26:
No one provided support so far, so nothing new? :)

Tasmota docs page is lacking an explanation of *how* it works. All key points are in the source, in comparison to the ifan2:
- instead of GPIO button presses, we need to read Serial.
https://github.com/arendst/Tasmota/blob/5014fb985b741002d1ccf9de23a872f33fe8e06c/tasmota/xdrv_22_sonoff_ifan.ino#L167-L175
and ACK for every data frame:
https://github.com/arendst/Tasmota/blob/5014fb985b741002d1ccf9de23a872f33fe8e06c/tasmota/xdrv_22_sonoff_ifan.ino#L147-L154
- fan speed 'GPIO masking' is slightly different for the highest speed, but that is still the same
https://github.com/arendst/Tasmota/blob/5014fb985b741002d1ccf9de23a872f33fe8e06c/tasmota/xdrv_22_sonoff_ifan.ino#L61-L66
But, I am not sure I agree with the current relayStatus() usage for this purpose :/ If we know speeds (aka 1,2,3,4), might as well have a setter for those, we don't need relays exposed at all and just manually do digitalWrite()s

davebuk, 2021-05-31:
I'm thinking of getting an iFan03 device. Did anyone get one working? I can't see any mention of it in `hardware.h`. If it can be supported to include the RF control as well as motor speed and light control via MQTT etc, I may look to buy one and help with any testing.

I currently have an IR only electronic control in a fan and also allows UP/DOWN dimming of the light. The iFan03 seems to just be ON/OFF for the light circuit but that shouldn't be too much off an issue if dimming is not supported.

mcspr, 2021-06-01:
[Original ifan support was reworked some time ago](https://github.com/xoseperez/espurna/commits/dev/code/espurna/ifan.cpp), but ifan03 is not (yet?) supported. 2nd bullet-point above is solved by having only one relay and a separate speed control, only the 1st one left - implement serial protocol to acknowledge the remote input

Revisiting the links:
https://github.com/arendst/Tasmota/blob/e59ce3da97b0aa832953310223635493071ed2af/tasmota/xdrv_22_sonoff_ifan.ino#L167-L176
there is at least one new thing - 'Rf long press - forget RF codes' (supposedly, related to the RF chip forgetting the specific remote, and not something we should handle)

For the outputs - it's GPIO9 for OFF / ON, GPIO14 for LOW, GPIO12 for MEDIUM and GPIO15 for HIGH
https://templates.blakadder.com/sonoff_ifan03.html

davebuk, 2021-06-02:
Sounds like a possibility to get it to work. If I get one I'll revisit the thread. Thanks.