xoseperez/espurna #1025: Add some useful feature (RTC, custom VСС, RELAY,BUTTONS)

poulch74, 2018-07-04:
Due process of adaptation my custom esp8266 board for this firmware I made some useful (i think) modificatoins. Maybe some of them can be integrated in upstream?

My custom board repo(not complete) https://github.com/poulch74/SHAPEsp
Forked espurna repo with modifications https://github.com/poulch74/espurna

Added modifications

[NTP] support for RTC ds3231 local time synchronization and timekeeping. Allow devices to function without internet connection with SCHEDULER capability (in STA and AP mode). With recovery attempts option to switch back for NTP time. see ntp.ino rtc3231.ino and RTC3231_SUPPORT defs...

[CORE] add support for custom VCC monitoring. See utils.ino int custom_getVcc(int mode) (in my device i use it for dc power (12V) monitoring before stepdown)

[RELAY] rework/rewrite initialization. add H-bridge(TB6612) relay, add ButtonReset relay etc. see RELAY folder. add binding for btnID to reset ONCE pushed button to it initial state.

[debounceevent] add ONCE mode for singleshot button. example usage - water sensor DO pin act like pushbutton and signal relay to close. with software relay ButtonReset we can manually reset back waterstop system. see DEBOUNCEEVENT\SRC folder

xoseperez, 2018-07-04:
What, such a bunch of goodies!
Sure I'd like to see some of these -if not all- into the main repo...
Would you like to do PR? Or maybe I should cherry-pick each functionality?

Misiu, 2018-07-05:
@poulch74 RTC support would be awesome.
I've noticed that You are using RTC3231, can DS1307 be also supported?
https://www.aliexpress.com/item/Integrated-Circuits-DataLog-Shield-for-WeMos-D1-mini-RTC-DS1307-with-battery-MicroSD-with-pin-headers/32818697084.html

poulch74, 2018-07-05:
I am trying to split my changes and make  some small PR in dev. But debounceevent for example not hosted on github and not a part of espurna, so some changes not work without PR to it on bitbucket... And i am trying add more complex support for RTC (ds1307) in few days, maybe week.