xoseperez/espurna #1026: Activating SoftAP on devices without any buttons?

jackwilsdon, 2018-07-04:
I've got a [Magic Home LED Controller](https://github.com/xoseperez/espurna/wiki/Hardware-Magic-Home-LED-Controller) but it's configured for the wrong WiFi - how do I put it in AP mode to reconfigure it?

ColinShorts, 2018-07-04:
Temporarily change the WiFi password or SSID on your router/access point, or switch router/AP off. Possibly then need to power cycle the espurna device.

jackwilsdon, 2018-07-04:
I've tried that and the device seems to _sometimes_ create an AP but it appears to be unreliable and disappears after a minute or two.

xoseperez, 2018-07-10:
The hack-ish way is to trigger safe-mode by powering it on and off 5 times in a raw. When this happens the devices considers there is a bug in any of the non-critical modules and it will boot into safe mode, only WiFIi AP, telnet and OTA available. You can then connect to the AP, telnet to it and reconfigure it.

closed 2018-09-08 by jackwilsdon