xoseperez/espurna #1030: Downgrade from 1.13 to 1.12 issue

ccapublic, 2018-07-05:
I upgraded a sonoff basic from latest 1.12 to 1.13. After this my jeedom could not control the sonoff any more. After investigation, I believe it is due to my sonoff hostname containing underscores... this was accepted under 1.12, but not in 1.13. The webpage under 1.13 shows the  hostname with undescores as inherited from 1.12 but clicking save without even changing a single setting causes espurna to complain about underscores. Because I can not easily/quickly change sonoff topic in jeedom, I decided to revert to 1.12. After downgrade, sonoff wont connect to my wifi any more. A hardware reset actually appear to restore defaults, I get a blinking LED, but AP does not appear... I guess I'm good for reflashing the device, am I ? It would be great for the future to avoid breaking backwards compatibility, or at least document it. That being said, this is a great FW ! Thanks for it !

xoseperez, 2018-07-10:
Previous hostname "rules" (actually no rules except for length) were not RFC compliant. This change was introduced to force people to use hostnames that will work with any external service, command, tool,... not doing it was actually creating some issues (see #874). 

I'm sorry it broke your setup.

closed 2018-07-10 by xoseperez

ccapublic, 2018-07-10:
Hi and thanks for your reply. What was most "annoying" was that downgrade to 1.12 appeared to completely mess up the fw. Luckily I tested 1.13 on an easily accessible BASIC so I flashed 1.13 again and simply forced MQTT topic to match the old hostname with underscores and everything was back to normal with jeedom. Thanks again. Cyril.