xoseperez/espurna #1033: Generic use of I2C ADC extenders liek ADS1115, ADC121

hyteoo, 2018-07-06:
Although there is support for I2C ADC extenders, they are implemented for energy measurement.

Would it be a good idea to just have like a generic ADS1x15 with raw outputs (like the internal ESP8266 ADC), to be used for whatever (voltage measurement, NTC, etc)?

flashy02, 2018-11-03:
Hi, the same issue here. I've bought an ADS1115 to use it as 4 Channel analog Input like the A0 from the ESP8266. It would be nice to get this feature. Thanks.

xoseperez, 2018-12-13:
See also #199 and #1311

mcspr, 2021-06-23:
As referenced in the commit, there is now EmonADS1X115Sensor::I2CPort:
https://github.com/xoseperez/espurna/blob/44298d4077ffd146c61eea7424b394167093a5d0/code/espurna/sensor.cpp#L1871-L1872
where `I2CPort::read(N)` is effectively an `analogRead(N)`

Only noticed the issue after the fact though. I guess it could be further modified as something virtual once again so there's a generic `virtual unsigned int Class::analogRead(int channel);`, but not inheriting from the BaseSensor stuff.