xoseperez/espurna #1037: OTA flashing of Sonoff S26 doesn't work

akosmaroy, 2018-07-06:
Hi,

I'm trying to flash a Sonoff S26 device according to the OTA flashing process described here: https://github.com/xoseperez/espurna/wiki/OTA-flashing-of-virgin-Itead-Sonoff-devices

but the process fails with the following, and the 'FinalStage' SSID never comes up.

here is my script output:

```
$ ./sonota.py --wifi-ssid XXX --wifi-password YYY --serving-host 192.168.30.102
Current IPs: ['192.168.30.102']

Using the following configuration:
	Server IP Address: 192.168.30.102
	WiFi SSID: XXX
	WiFi Password: **********
Platform: linux
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
..........................Current IPs: []
.Current IPs: ['10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100046605d",
    "apikey": "29fba2a0-2bb9-4e63-bde3-417c42a67eaf",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "XXX",
    "password": "**********",
    "serverName": "192.168.30.102",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.30.102) is not assigned to any interface on this machine.
** Please change WiFi network to HappyMiner and make sure 192.168.30.102 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
.........Current IPs: []
.Current IPs: ['192.168.30.102']
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
Using selector: EpollSelector
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
...........Uncaught exception, closing connection.
Traceback (most recent call last):
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 657, in _handle_events
    self._handle_read()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 1068, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 689, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
Exception in callback None()
handle: <Handle cancelled>
Traceback (most recent call last):
  File "/usr/lib/python3.6/asyncio/events.py", line 145, in _run
    self._callback(*self._args)
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/platform/asyncio.py", line 117, in _handle_events
    handler_func(fileobj, events)
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/stack_context.py", line 276, in null_wrapper
    return fn(*args, **kwargs)
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 657, in _handle_events
    self._handle_read()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 1068, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 689, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
...........Uncaught exception, closing connection.
Traceback (most recent call last):
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 657, in _handle_events
    self._handle_read()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 1068, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 689, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
Exception in callback None()
handle: <Handle cancelled>
Traceback (most recent call last):
  File "/usr/lib/python3.6/asyncio/events.py", line 145, in _run
    self._callback(*self._args)
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/platform/asyncio.py", line 117, in _handle_events
    handler_func(fileobj, events)
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/stack_context.py", line 276, in null_wrapper
    return fn(*args, **kwargs)
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 657, in _handle_events
    self._handle_read()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 1068, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 689, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
........
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
...Uncaught exception, closing connection.
Traceback (most recent call last):
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 657, in _handle_events
    self._handle_read()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 1068, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 689, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
Exception in callback None()
handle: <Handle cancelled>
Traceback (most recent call last):
  File "/usr/lib/python3.6/asyncio/events.py", line 145, in _run
    self._callback(*self._args)
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/platform/asyncio.py", line 117, in _handle_events
    handler_func(fileobj, events)
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/stack_context.py", line 276, in null_wrapper
    return fn(*args, **kwargs)
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 657, in _handle_events
    self._handle_read()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "/home/akos/.local/lib/python3.6/site-packages/tornado/iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 1068, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 689, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error

```

I wonder what I'm doing wrong

Akos

xoseperez, 2018-07-10:
I can't help. The S26 is not in the list of known-working devices, maybe someone else can help...

closed 2018-09-15 by stale[bot]

rzr, 2018-10-21:
I have a S26 but I dont see the "ITEAD-100001xxxx" showing when in AP mode ? Is it an hidden SSID ?