xoseperez/espurna #1038: Sonoff POW not showing Power/Current/Voltage

stelgenhof, 2018-07-07:
My Sonoff POW using Espurna 1.13.0 doesn't show any active Power/Current/Voltage (in the status screen).

As for the calibration (Sensors screen), when I enter the expected Power/Current/Voltage the numbers are reset to zero after pressing Save.

I've tried also version 1.12.5, 1.12.6 and the current dev version, but no success. Everything else works fine (MQTT is connected and the switch works fine as well).

Is there something I can check?

ServerCare, 2018-07-09:
what is the load you have attached to it?

stelgenhof, 2018-07-09:
I tried with different loads. First an LED strip with power adapter. Measured 101V, 0.53A (about 36W).
Second time a more higher load 102V, 2.1A (About 200W).

(I am in Japan where AC is 100V).

xoseperez, 2018-07-09:
@stelgenhof is it reporting correct values via MQTT or they are also set to zero? The values in the "expected" fields are always reset to zero, that is OK (they are current expected values, not calibration values).

xoseperez, 2018-07-09:
Wait, by default, if the relay is OFF the reported values will be 0. Check if it works when the relay in ON.

stelgenhof, 2018-07-10:
@xoseperez The relay is ON (See screenshot). Also added a screenshot of the DEBUG console showing MQTT is reporting however all values are zero.

[image]
[image]

zerog2k, 2018-07-14:
FWIW, I was having some difficulty calibrating on a KMC 70011 a few months back. I just upgraded to 1.13.1 and did calibration. Not sure if this was an issue, but when entering expected (current/voltage/power) fields, do _not_ do "reset calibration". I think this does a separate reset of calibration data to factory defaults, which is _not_ what I wanted here, although I think the UI confused me a little bit.

(If this is indeed the case, then there could probably be a horizontal separator between the `expected` fields and the `reset calibration` switch to reinforce that these are mutually exclusive operations.)

stelgenhof, 2018-07-15:
Thanks! I didn't do the reset calibration, however just wanted to add the calibration values (Expected power/current/voltage). Not sure then if I understand how the UI works. When I enter a value for expected power and press Save, the entered value gets set to zero (on screen). Maybe it gets stored in the background...

zerog2k, 2018-07-15:
Yeah, it seems to happen for me as well. Try putting your known pure resistive load on.
For me, I had a kill-a-watt style watt meter which I could look at the current, voltage, and power to get approximate values. Then enter all three of the expected current, voltage, and power, and save.
Go back to status screen to see if the values reported are different now (influenced by calibration settings)

jeanpaul79, 2018-07-15:
Hello,

I have the same problem. If switch is on, all the values are at 0. If switch is off I only get the voltage value.
I did the calibration but it didn't change anything. I have 1.13.1 from github.

xoseperez, 2018-07-16:
The values in the web UI are not calibration values. Instead, they are **expected** values, and thus they depend on the moment. You have to connect a known load and enter the expected values for that load. The firmware will calculate the calibration values based on the current calibration values, the current calculated value and the expected value you provide. The calibration values will be stored in the EEPROM and accessible via terminal under `pwrRatioC`, `pwrRatioV` and `pwrRatioP`.
Once the expected values are used they are reset to 0.

jeanpaul79, 2018-07-17:
Hello,

Yes on the "sensors" page. But in the status all the values (power, current, voltage) are at 0. If I turn the switch off, only voltage is displayed.

My calibration values: 
> pwrRatioC => "14484.49"
> pwrRatioV => "408636.51"
> pwrRatioP => "10343611.74"

xoseperez, 2018-07-23:
Calibration values look OK. When switch is OFF it will always report voltage only. But when ON it should report all values, of course. Is it a POW or a POW R2?

jeanpaul79, 2018-07-24:
Ok thank you. It is a POW

When OFF it behaves at it should but when ON it displays only 0 everywhere.

xoseperez, 2018-07-26:
Can you enable SENSOR_DEBUG in sensors.h and compile and flash the image to your device? It will start logging more info about the communications from the CSE7766 that will help us find out what's going on...

jeanpaul79, 2018-07-28:
Hello, here are the results (V1.13.1)

With relay OFF
 `[736624] [SENSOR] HLW8012 @ GPIO(5,14,13) - current: 0.000A
[736628] [SENSOR] HLW8012 @ GPIO(5,14,13) - voltage: 218V
[736630] [SENSOR] HLW8012 @ GPIO(5,14,13) - power: 0W
[736631] [SENSOR] HLW8012 @ GPIO(5,14,13) - reactive: 0W
[736635] [SENSOR] HLW8012 @ GPIO(5,14,13) - apparent: 0W
[736643] [SENSOR] HLW8012 @ GPIO(5,14,13) - factor: 0%
[736646] [SENSOR] HLW8012 @ GPIO(5,14,13) - energy: 217J`

With relay ON
`[766659] [SENSOR] HLW8012 @ GPIO(5,14,13) - current: 0.000A
[766662] [SENSOR] HLW8012 @ GPIO(5,14,13) - voltage: 0V
[766664] [SENSOR] HLW8012 @ GPIO(5,14,13) - power: 0W
[766666] [SENSOR] HLW8012 @ GPIO(5,14,13) - reactive: 0W
[766671] [SENSOR] HLW8012 @ GPIO(5,14,13) - apparent: 0W
[766676] [SENSOR] HLW8012 @ GPIO(5,14,13) - factor: 0%
[766679] [SENSOR] HLW8012 @ GPIO(5,14,13) - energy: 222J`

jeanpaul79, 2018-08-05:
Hello,

with 1.12.4 it's working fine. There're some gliches, sometimes it reads 0 but otherwise it's OK.
with 1.12.5 it doesn't work. It shows voltage only when the switch is off and when switch is on, everything is at 0.

I tried 1.13.1 with SENSOR_POWER_CHECK_STATUS set to 0 but it's the same.

I did a comparison https://github.com/xoseperez/espurna/compare/1.12.4...1.12.5 but I don't manage to find where the problem come from.

adityama, 2019-05-12:
I am facing same issue with PZEM004T module. The module (and wiring etc) were verified with Sonoff Tasmota firmware.  With ESPURNA firmware ver 1.13.6-dev it fails to read "Current" and "Power". The sample outputs are below -

[581096] [MQTT] Sending UPSpwMon/current/0 => 0.000 (PID 1)
[581099] [MQTT] Sending UPSpwMon/voltage/0 => 240 (PID 1)
[581101] [MQTT] Sending UPSpwMon/power/0 => 0 (PID 1)
[581145] [MQTT] Sending UPSpwMon/energy/0 => 1.286 (PID 1)
[581208] [MQTT] Sending UPSpwMon/current/1 => 0.000 (PID 1)
[581268] [MQTT] Sending UPSpwMon/voltage/1 => 240 (PID 1)
[581332] [MQTT] Sending UPSpwMon/power/1 => 0 (PID 1)
[581386] [MQTT] Sending UPSpwMon/energy/1 => 1.191 (PID 1)
[641136] [MQTT] Sending UPSpwMon/current/0 => 0.000 (PID 1)
[641139] [MQTT] Sending UPSpwMon/voltage/0 => 240 (PID 1)
[641141] [MQTT] Sending UPSpwMon/power/0 => 0 (PID 1)
[641186] [MQTT] Sending UPSpwMon/energy/0 => 1.293 (PID 1)
[641249] [MQTT] Sending UPSpwMon/current/1 => 0.000 (PID 1)
[641309] [MQTT] Sending UPSpwMon/voltage/1 => 240 (PID 1)
[641368] [MQTT] Sending UPSpwMon/power/1 => 0 (PID 1)
[641428] [MQTT] Sending UPSpwMon/energy/1 => 1.197 (PID 1)

I have tried this with single PZEM module as well, the problem remains same. Any suggestions?

adityama, 2019-05-12:
I gave it a shot with with SENSOR_POWER_CHECK_STATUS set to 0 and it worked fine -

[079217] [MQTT] Sending UPSpwMon/current/0 => 2.125 (PID 1)
[079220] [MQTT] Sending UPSpwMon/voltage/0 => 236 (PID 1)
[079222] [MQTT] Sending UPSpwMon/power/0 => 477 (PID 1)
[079271] [MQTT] Sending UPSpwMon/energy/0 => 2.296 (PID 1)
[079332] [MQTT] Sending UPSpwMon/current/1 => 2.020 (PID 1)
[079395] [MQTT] Sending UPSpwMon/voltage/1 => 235 (PID 1)
[079455] [MQTT] Sending UPSpwMon/power/1 => 449 (PID 1)
[079514] [MQTT] Sending UPSpwMon/energy/1 => 2.132 (PID 1)
[139242] [MQTT] Sending UPSpwMon/current/0 => 2.130 (PID 1)
[139246] [MQTT] Sending UPSpwMon/voltage/0 => 236 (PID 1)
[139248] [MQTT] Sending UPSpwMon/power/0 => 476 (PID 1)
[139294] [MQTT] Sending UPSpwMon/energy/0 => 2.305 (PID 1)
[139356] [MQTT] Sending UPSpwMon/current/1 => 2.022 (PID 1)
[139419] [MQTT] Sending UPSpwMon/voltage/1 => 236 (PID 1)
[139479] [MQTT] Sending UPSpwMon/power/1 => 448 (PID 1)
[139537] [MQTT] Sending UPSpwMon/energy/1 => 2.140 (PID 1)

ruimarinho, 2019-05-12:
@xoseperez I am experiencing a somewhat similar issue - using a Blitzwolf BW-SHP2 (original board revision) I always get 0W reading from the power sensor. Sometimes when I quickly turn off and on the relay, I get an instant reading and then it drops to 0W throughout that cycle.

I was on 1.13.3 and upgraded to 1.13.5 but no changes on this particular issue.

mcspr, 2019-05-13:
@adityama Just to verify - does that mean that you have a single relay configured?
https://github.com/xoseperez/espurna/blob/55b9f2a52f9fc5a225ae4880d15893d15161209f/code/espurna/sensor.ino#L1473-L1475
(based on this check)

pcornill, 2020-12-07:
Hello, I have the same issue. I just upgraded from Espurna 1.14.2 to 1.15.0 and all measurements except Voltage are set to zero; even when the switch is ON and loaded

mcspr, 2020-12-07:
@pcornill can you be more specific about the version? which commit is that? does it show values after power-cycling the device?

pcornill, 2020-12-07:
It is version 1.15.0-dev. And as I said "all measurements except Voltage are set to zero"
[image]

mcspr, 2020-12-07:
You will see the 'revision' field on the same page, near the version.
Have you tried powering off the device or resetting to factory settings? I mean it. https://gitter.im/tinkerman-cat/espurna?at=5fac3135d37a1a13d69e1993

pcornill, 2020-12-08:
I tried powering off but I do not know how to upload OTA and erase_all (I do not have easy access to serial upload)

[image]

mcspr, 2020-12-09:
> OTA and erase_all (I do not have easy access to serial upload)

I meant the "factory reset" button, which will remove the active settings. Simply flashing the new OTA won't do it, settings will be preserved. Or, just send `factory.reset` in the DEBUG command input.

The gitter discussion meant to point out it is possible to sort-of break display values when setting 'expected' measurements through the combination of expected settings and build defaults, which was fixed ~month ago

Due to the missing 'revision', I assume it's a custom build? What is the specific commit that you used?