xoseperez/espurna #1039: How to configure a relay with turn-high-low via GPIO?

Yonsm, 2018-07-08:
Hi Xoseperez, I want to configure a relay that represents a high-low voltage  at GPIO for both turn on/off action, that means I need toggle GPIO pin to high and then turn GPIO pin to low (about 0.3s delay) for both turn-on and turn-off action.

I found `type(RELAY_TYPE_LATCHED)/delay_on/delay_off/pulse/pulse_ms` params maybe is the right way, but I did not find relay config UI on web frontend. How can I get there, or how can config the relay on `hardware.h`.

Thank you and your help will be highly appreciated:)

Yonsm, 2018-07-08:
I found that!  The following configuration could be worked well:

```
    #define RELAY1_PIN          14
    #define RELAY1_RESET_PIN    14
    #define RELAY1_TYPE         RELAY_TYPE_LATCHED
    #define RELAY_LATCHING_PULSE 200
```

But the best code for me is (no reset_pin needed):
```
# relay.ino #117 add a new RELAY_TYPE_STRIKE
digitalWrite(_relays[id].pin, !pulse);
digitalWrite(_relays[id].pin, pulse);
nice_delay(RELAY_LATCHING_PULSE);
digitalWrite(_relays[id].pin, !pulse);
```

If you can support this relay mode, it's a good news:)

xoseperez, 2018-07-10:
Both codes are equivalent. Instead of creating a new relay type I prefer to support single pin latched relays using the same type. I've added support for it in the dev branch. Configuration will be like this:

```
    #define RELAY1_PIN          14
    #define RELAY1_TYPE         RELAY_TYPE_LATCHED
    #define RELAY_LATCHING_PULSE 200
```

(no RELAY1_RESET_PIN)

closed 2018-07-10 by xoseperez

Yonsm, 2018-07-14:
Well done. Thanks a lot!