xoseperez/espurna #1040: How to use Sensors

AHBkingman, 2018-07-08:
Hello Friends,
Do anybody have an Idea on, how to use Power and Energy meter sensors with Sonoff devices?

- With which Sonoff device?
- How to connect?
- How to configure?

ceaswaran, 2018-07-09:
I use 433Mhz PIR (Motion sensor) with Sonoff RF bridge 
I have flashed Sonoff RF Bridge with Espurana Firmware.
Is that you are looking for?

xoseperez, 2018-07-10:
Hi @AHBkingman, there are a couple of Sonoff devices that support power metering: the POW and the POW R2. They use different metering chips but they are both supported by ESPurna. You just have to flash the corresponding image to get it working.

If you want to use a custom power meter with a different Sonoff device (or any other device) you will have to manually enable support for the specific power meter method. At the moment ESPurna supports:

* Direct current monitor using a CT connected to A0
* Current monitor using a CT connected to an ACS712 connected to A0
* Current monitor using a CT connected to an ADC121
* Current monitor using a CT connected to an ADS1117
* HLW8012 power monitor chip (the one in the Sonoff POW)
* CSE7766 power monitor chip (the one in the Sonoff POW R2, some variants also supported)
* HJL-01 and BL0937 power monitor chip
* V9217F power monitor chip (available in several cheap Chinese wall monitors)
* PZEM0004T power monitor board

AHBkingman, 2018-07-12:
Thank you Xoseperez for your useful comment,
I am specifically interested to connect a CT current sensor to Sonoff TH to monitor Current, Voltage and Energy consumption for an AC Electric Lines with over 50 A current with 220V. 
I understand that the best way to do it is by using a CT noninvasive current Sensor, like the one you had used in your [post with Sonoff TH and ADC121](http://tinkerman.cat/power-monitoring-sonoff-th-adc121/). However, if it is possible to connect the CT sensor directly it will be a great step.
Please, can you refer any example or reference to apply the configuration and connections?

xoseperez, 2018-07-12:
With a CT sensor you can only monitor current and apparent power, not voltage or active/real power.

A CT sensor also requires a minimum circuitry: a voltage shifter and a filter or an opamp. And it outputs an analog value so you will have to wire it to the TOUT pin in the ESP8288, which is not in the TH external connector. This is quite tricky, but doable (check this: http://tinkerman.cat/your-laundry-is-done/).

The post you link above is a solution to bypass the lack of analog input in the 2.5 TRRS connector and include the needed circuit to read the CT.

AHBkingman, 2018-07-12:
What changes should I modify if I want to use [ADS1115 16-bit Module found on AliExpress](https://www.aliexpress.com/item/16-Bit-I2C-ADS1115-Module-ADC-4-channel-with-Pro-Gain-Amplifier-for-Arduino-RPi-1PCS/32817162654.html?spm=2114.search0104.3.1.773077b9iAexT4&ws_ab_test=searchweb0_0,searchweb201602_5_10846_10152_10065_10151_10344_10068_10342_10343_10340_10059_10341_10696_100031_10084_10083_10103_524_10618_10624_10307_10623_10622_10621_10620,searchweb201603_6,ppcSwitch_5&algo_expid=887a96e0-ad48-4e1e-a2e3-350b4d9b9551-0&algo_pvid=887a96e0-ad48-4e1e-a2e3-350b4d9b9551&priceBeautifyAB=0) instead of using the [ADC121 module  in your post](http://tinkerman.cat/power-monitoring-sonoff-th-adc121/).

xoseperez, 2018-07-17:
ESPurna supports both ADC for power monitoring, simply use the `EmonADS1X15Sensor` sensor class instead of the `EmonADC121Sensor` class.

closed 2018-09-22 by stale[bot]