xoseperez/espurna #1047: YJZK 3 gang support

q32103940, 2018-07-12:
I noticed only the 2-gang switch is supported. Will 3-gang support for this [T1 clone](https://item.taobao.com/item.htm?spm=a1z0d.6639537.1997196601.15.2a3874849q4d7g&id=554114857768) be possible?

ColinShorts, 2018-07-13:
If it is more or less the same then it should be possible. There are some extra lines in the SONOFF T1 defines, and LED_PIN_INVERSE is different.

Do you already have a 3-gang (and/or 1-gang) and are willing to test it?

q32103940, 2018-07-17:
Hi Xose i'm willing to test my 3-gang (I only have this). I've loaded Sonoff T1 3ch template. however, the middle switch led (#2) lights up when switch #1 or #3 are pressed (i.e., when you press #1, #1 & #2 light up. when you press #3, #1 and #3 light up). apart from the switch#2 LED behavior, the relays seem to be matched correctly (#1,#2,#3 are GPIO0,9,10 respectively). i was able to surpress the weird LED behavior of #2 by forcing LED_PIN_INVERSE to 1 but i dont think this is the perfect solution

q32103940, 2018-07-17:
please let me know how i can help to get this template added!

devel-intrallect, 2018-07-17:
You've not said what you are using to build (platformio / arduino), but I suspect that the configs will closely mirror that of the Sonoff's as per my previous suggestion. There are 4 files that should change to support new devices, and I've knocked together what I would insert into each of them below for the 1/2/3 channel devices:

code/espurna/config/arduino.h:
```
...
//#define YJZK_SWITCH_1CH
//#define YJZK_SWITCH_2CH
//#define YJZK_SWITCH_3CH
...
```
code/espurna/config/hardware.h:
```
...
#elif defined(YJZK_SWITCH_1CH)

    // Info
    #define MANUFACTURER        "YJZK"
    #define DEVICE              "SWITCH_1CH"

    // Buttons
    #define BUTTON1_PIN         0

    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          12

    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    0

#elif defined(YJZK_SWITCH_2CH)

    // Info
    #define MANUFACTURER        "YJZK"
    #define DEVICE              "SWITCH_2CH"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON2_PIN         9

    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    #define BUTTON1_RELAY       1
    #define BUTTON2_RELAY       2

    // Relays
    #define RELAY1_PIN          12
    #define RELAY2_PIN          5

    #define RELAY1_TYPE         RELAY_TYPE_NORMAL
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    0

#elif defined(YJZK_SWITCH_3CH)

    // Info
    #define MANUFACTURER        "YJZK"
    #define DEVICE              "SWITCH_3CH"

    // Buttons
    #define BUTTON1_PIN         0
[... 127 lines cut ...]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_1m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_1m0m} -DYJZK_SWITCH_3CH
monitor_speed = ${common.monitor_speed}
extra_scripts = ${common.extra_scripts}

[env:yjzk-switch-3ch-ota]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_1m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_1m0m} -DYJZK_SWITCH_3CH
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
upload_flags = ${common.upload_flags}
monitor_speed = ${common.monitor_speed}
extra_scripts = ${common.extra_scripts}
...
```

devel-intrallect, 2018-07-17:
NOTE: I would take @xoseperez advice on whether to include the 1CH in any pull request - ideally it should be tested first.

ColinShorts, 2018-07-17:
You've not said what you are using to build (platformio / arduino), but I suspect that the configs will closely mirror that of the Sonoff's as per my previous suggestion. There are 4 files that should change to support new devices, and I've knocked together what I would insert into each of them below for the 1/2/3 channel devices:

code/espurna/config/arduino.h:
```
...
//#define YJZK_SWITCH_1CH
//#define YJZK_SWITCH_2CH
//#define YJZK_SWITCH_3CH
...
```
code/espurna/config/hardware.h:
```
...
#elif defined(YJZK_SWITCH_1CH)

    // Info
    #define MANUFACTURER        "YJZK"
    #define DEVICE              "SWITCH_1CH"

    // Buttons
    #define BUTTON1_PIN         0

    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          12

    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    0

#elif defined(YJZK_SWITCH_2CH)

    // Info
    #define MANUFACTURER        "YJZK"
    #define DEVICE              "SWITCH_2CH"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON2_PIN         9

    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    #define BUTTON1_RELAY       1
    #define BUTTON2_RELAY       2

    // Relays
    #define RELAY1_PIN          12
    #define RELAY2_PIN          5

    #define RELAY1_TYPE         RELAY_TYPE_NORMAL
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    0

#elif defined(YJZK_SWITCH_3CH)

    // Info
    #define MANUFACTURER        "YJZK"
    #define DEVICE              "SWITCH_3CH"

    // Buttons
    #define BUTTON1_PIN         0
[... 131 lines cut ...]
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_1m0m} -DYJZK_SWITCH_3CH
monitor_speed = ${common.monitor_speed}
extra_scripts = ${common.extra_scripts}

[env:yjzk-switch-3ch-ota]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_1m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_1m0m} -DYJZK_SWITCH_3CH
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
upload_flags = ${common.upload_flags}
monitor_speed = ${common.monitor_speed}
extra_scripts = ${common.extra_scripts}
...
```

NOTE: I would take @xoseperez advice on whether to include the 1CH in any pull request - ideally it should be tested first.

EDIT: just noticed I was logged in with the wrong account, sorry.

xoseperez, 2018-07-23:
@q32103940 Can you test @ColinShorts changes and confirm his config is working for the 3CH YJZK?

q32103940, 2018-07-26:
sorry for the delay, I have been busy at work and couldnt get back on this project. i will report back in the next two weeks.

CrazyIvan359, 2018-07-26:
This sounds like the switch I just reverse engineered myself. If it is [this device](https://www.aliexpress.com/item/New-Ewelink-US-Type-1-2-3-gang-wall-light-app-switch-touch-control-panel-wifi/32845778982.html?spm=a2g0s.9042311.0.0.12154c4dSODQnJ) then here is a device profile that will work with any number of switches. I currently only have a 1 switch device that I have tested, but I mapped the whole board and am confident that this will work with the 2 and 3 switch devices as well.

There are unpopulated footprints for what I believe is a 433MHz radio that looks to be connected to the relay controls. There are also unpopulated footprints for another radio that is hooked up to the HD6001 touch pad controller. I have not looked into them further, as there is no indication of what chip goes there.

There is also a second footprint for a Wifi controller, I assume the install one or the other. My board has a PSF-B01 (PSF-B85) installed on it and this configuration is for that chip. I can also provide a configuration for the other board, the footprint looks like that of an ESP-12x.

I copied the button configuration from the Sonoff Touch, because touch buttons respond differently than the physical buttons Espurna expects.

```
#elif defined(LX_WIFI_03M)

    // Info
    #define MANUFACTURER        "LIXIN"
    #define DEVICE              "LIXIN_TOUCH"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_PRESS       BUTTON_MODE_TOGGLE
    #define BUTTON1_CLICK       BUTTON_MODE_NONE
    #define BUTTON1_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON1_LNGCLICK    BUTTON_MODE_NONE
    #define BUTTON1_LNGLNGCLICK BUTTON_MODE_RESET
    #define BUTTON1_RELAY       1

    #define BUTTON2_PIN         9
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON2_PRESS       BUTTON_MODE_TOGGLE
    #define BUTTON2_CLICK       BUTTON_MODE_NONE
    #define BUTTON2_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON2_LNGCLICK    BUTTON_MODE_NONE
    #define BUTTON2_LNGLNGCLICK BUTTON_MODE_RESET
    #define BUTTON2_RELAY       2

    #define BUTTON3_PIN         10
    #define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON3_PRESS       BUTTON_MODE_TOGGLE
    #define BUTTON3_CLICK       BUTTON_MODE_NONE
    #define BUTTON3_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON3_LNGCLICK    BUTTON_MODE_NONE
    #define BUTTON3_LNGLNGCLICK BUTTON_MODE_RESET
    #define BUTTON3_RELAY       3

    // Relays
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL
    #define RELAY2_PIN          5
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL
    #define RELAY3_PIN          4
    #define RELAY3_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    1
```

q32103940, 2018-07-26:
@ColinShorts Hey Colin, I ended up using platformio. I ended up getting it working.

> code/espurna/migrate.h:

Couldn't find this file, so I assume you meant migrate.ino

For hardware.h, I used this below. There were typos related to BUTTON2 on the snippet you pasted above. After fixing that the thing worked like magic.

@xoseperez this section worked for code/espurna/config/hardware.h:

>     #elif defined(YJZK_SWITCH_3CH)
> 
>     // Info
>     #define MANUFACTURER        "YJZK"
>     #define DEVICE              "SWITCH_3CH"
> 
>     // Buttons
>     #define BUTTON1_PIN         0
>     #define BUTTON2_PIN         9
>     #define BUTTON3_PIN         10
>     
>     #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
>     #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
>     #define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
> 
>     #define BUTTON1_RELAY       1
>     #define BUTTON2_RELAY       2
>     #define BUTTON3_RELAY       3
> 
>     // Relays
>     #define RELAY1_PIN          12
>     #define RELAY2_PIN          5
>     #define RELAY3_PIN          4
> 
>     #define RELAY1_TYPE         RELAY_TYPE_NORMAL
>     #define RELAY2_TYPE         RELAY_TYPE_NORMAL
>     #define RELAY3_TYPE         RELAY_TYPE_NORMAL
>     
>     // LEDs
>     #define LED1_PIN            13
>     #define LED1_PIN_INVERSE    0

@CrazyIvan359 thanks for joining the discussion and providing your template. seems like the YJZK doesn't need the LED_PIN1_INVERSE=1. just curious, how would your switch behave if it was =0 ?

CrazyIvan359, 2018-07-26:
@q32103940 I just saw the link posted to the device, looks like you have an EU version? The one I have is the US/CAN version and the WiFi LED behaves correctly with `LED_PIN1_INVERSE 1`. Interesting, since all of the pins are the same but the LED signal is inverted.

How do the buttons behave with the normal settings like you have configured? I flashed mine with the copied Sonoff Touch button settings and I like how it works, as soon as you make contact with the touch pad the switch changes.

q32103940, 2018-07-27:
@CrazyIvan359 

mine's the UK version. my switches only activate on release not on press. is that what LED_PIN1_INVERSE toggles?

CrazyIvan359, 2018-07-27:
@q32103940 
No, the following additional options for the buttons are what makes it activate on press instead of release. I find for this device it's more intuitive.

```
    #define BUTTON1_PRESS       BUTTON_MODE_TOGGLE
    #define BUTTON1_CLICK       BUTTON_MODE_NONE
    #define BUTTON1_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON1_LNGCLICK    BUTTON_MODE_NONE
    #define BUTTON1_LNGLNGCLICK BUTTON_MODE_RESET
```
I also just noticed that I have `#define BUTTON1_LNGLNGCLICK BUTTON_MODE_RESET` set for all buttons, I should change that to `#define BUTTON1_LNGLNGCLICK BUTTON_MODE_NONE` for buttons 2 and 3.

q32103940, 2018-07-27:
@CrazyIvan359 
i see now there is distinction between PRESS and CLICK. i might go that route. i have two questions...how does your switch behave if you used LED1_PIN_INVERSE   1, and unrelated, can we have click=toggle and dblclick=pulse mode?

CrazyIvan359, 2018-07-27:
@q32103940 
If I were to use `LED1_PIN_INVERSE 1`, my LED would be on when Espurna commands it to be off and vice versa, that's all.

You should be able to use this to have click toggle and double click pulse:
```
    #define BUTTON1_CLICK       BUTTON_MODE_TOGGLE
    #define BUTTON1_DBLCLICK    BUTTON_MODE_PULSE
```
You won't be able to do this if you wanted to use `BUTTON1_PRESS` to have the device respond when you press the button. You will be limited to only using the button release event