xoseperez/espurna #1051: Multiple PZEM004T

0x3333, 2018-07-13:
Hi!

I'm working on a project that has 3 PZEM004T sensors connected to the same UART. They have addresses and only respond when the message is directed to them. 

My schematics is like this:
[image]

What is the way you think is right to implement on ESPurna? Create a new "sensor" , or retrofit the actual `PZEM004TSensor.h`?

xoseperez, 2018-07-17:
Not sure but I would first try to use the current PZEM004TSensor class to handle it. Currently, it has a hardcoded address of 192.168.1.1. So you will first need to be able to pass a certain value and also a way to set the addresses individually. Not sure how this is done...

0x3333, 2018-07-20:
Looks like it is working, just some small changes to the diff I sent.

I also added a command `PZEM.SETADDR X.X.X.X`, this will set the Address in the PZEM004T, so the user can disconnect all PZEMs and connect one by one to set its address. This is necessary only once, the PZEM will hold its address across restarts.

Unfortunately, I have only 2 PZEMs to test, but others in the mail.

[image]

0x3333, 2018-07-20:
I'll finish some tests and submit a PR.

0x3333, 2018-07-23:
Well, after some digging, looks like the ESP can't handle 3 or more devices, it hangs, probably too busy to handle all magnitudes of all devices. I made it work for 3 devices, but too much unreliable, some readings fail, sometimes the device panic, doesn't worth the PR. I'll work on a version using an Atmega Pro Mini handling the PZEM reading and forwarding to an ESP-01 module with Espurna, probably will create a new sensor that can handle all readings in the same request.

I'll close this bug.

0x3333, 2018-10-29:
I'm reopening this issue because I could figure out a way to handle multiple PZEM004T devices.

The concept is the same, multidrop UART TX.

I also added 3 commands:

`pz.address [ADDRESS]`: Print the addresses of all devices, with `ADDRESS` will issue `Set Address` command on the bus. This is to be used to program PZEM004T to a specific address.
`pz.reset [DEV_INDEX]`: Reset the Energy magnitude on all devices or the device index specified(`DEV_INDEX`).
`pz.value [DEV_INDEX]`:  Print the last reading of all devices or the device index specified(`DEV_INDEX`).

I just need to confirm that the set address in the PZEM004T is persistent across reboots, otherwise, I'll need to add more changes, 

I'll submit the PR to start the discussion.

closed 2018-12-11 by xoseperez