xoseperez/espurna #1052: Sonoff RF Bridge

jsponz, 2018-07-13:
Question: I have installed espurna on a Sonoff RF Bridge. I am using a door sensor (magnet) and a movement sensor. 

When I do the learning phase, I detect the ON code (door opened and movement), but there is no code for OFF (door closed and no movement). 

How can I get the OFF signal so the switch is off?

Thanks!

xoseperez, 2018-07-17:
I have not tested the magnet sensor with the RF Bridge. I guess you get the ON code when you open the door, but you are not getting any code when you close it? Could it be the same code (weird)?

jsponz, 2018-07-17:
Yes, you are right. No code received when closing. 

When opening I get a code (and also a blue LED blinks on the magnet sensor). 

Anyone with a working magnet sensor for doors in Sonoff RF Bridge?

hyteoo, 2018-07-17:
What magnet sensor are you using? Can you post a link to where you bought it?
I'm also waiting for some generic 2$ door sensors from ebay and also 5$ sonoffs. From what I've read the cheap ones only send a signal on open.
Will keep you posted when I get them

jsponz, 2018-07-17:
These are the ones:

https://www.amazon.es/inal%C3%A1mbrica-Detector-movimiento-inmunidad-vigilancia/dp/B0798Q1ZJ8/ref=sr_1_27?srs=10532928031&ie=UTF8&qid=1531832283&sr=8-27

jsponz, 2018-07-17:
@hyteoo when tested, please share your links if they work :-)

closed 2018-09-28 by stale[bot]

MihataBG, 2018-09-28:
@hyteoo
Can you share a link of sensor?