xoseperez/espurna #1053: Need Help 😔 - rx,tx,ioo,gnd positions

tomimarcincin, 2018-07-13:
Asking for help, i have new controller and dont know which pins are the right to connect programmer.. 

Here are the images of the board

[image]
[image]

AlbertWeterings, 2018-07-13:
It looks like yes. It might be rx&tx have to be switched hard to see on the picture. I would start with GND and V3.3 if the device starts working connect rx tx and hold IOo to GND during power up.

ColinShorts, 2018-07-13:
You're going to have to break out the soldering iron for tx and Rx, and you'll still need to ground gpio0 to get it into flash mode. You should be able to power it from the 3.3v and gnd headers, but if you're soldering the rest there's no harm in soldering the power wires too.

What module is it?

tomimarcincin, 2018-07-13:
Thank you guys :) its Mini Wifi RGB controller from Aliexpress ( https://www.aliexpress.com/item/RGB-strip-controller-By-Amazon-Alexa-Google-Home-Phone-WIFI-RGB-Controller-5M-10M-15M-20M/32855869283.html )

tomimarcincin, 2018-07-13:
[image]

I have this programmer 
Black ---- GND
Green --- TX
White --- RX
Red --- VCC (5V)

What steps to take  ? Can i do this procedure
1. connect GND or V33 to Red Cable ? or ?
2. RX to TX and TX to RX (this is easy)
3. Connect GPIO00 to GND (Black cable) During start ?

ColinShorts, 2018-07-13:
You need to use a 3.3v serial adapter, 5v will likely kill it. An ftdi clone would do the trick. If you have a voltmeter you can check the cable you have, but if it's higher than 3.5V don't use it.

Black is usually gnd, red would be +V / VCC.

Sometimes you need to swap the Rx and Tx lines to get it working.

tomimarcincin, 2018-07-13:
@ColinShorts 
AL-LC01 Working Voltage: DC 5-28V  
So i think 5V will not kill it or am i wrong ?

AlbertWeterings, 2018-07-13:
The esp chip is working on 3v3 So you Will fry it if you put 5v on the v3.3 point. In your device the supply voltage is decreased by a regulator that is most likely the little chip with 6 legs on the right side of the yellow capacitor.

ColinShorts, 2018-07-13:
No worries. That only really applies to devices that are powered directly by the mains.

If you are flashing devices you can normally power them via the serial cable, if you need to apply external power then you need to ensure that the serial cable and external power supplies ground (negative) wire are connected to the device. Don't connect the serial power in that case.

tomimarcincin, 2018-07-13:
Thank you :)  can i ask you where do i find espurna firmware for my broad after binary flashing ?

ColinShorts, 2018-07-13:
The firmware should be a one step flash, the release files are here:
 https://github.com/xoseperez/espurna/releases

tomimarcincin, 2018-07-13:
And after reboot ill be able to access Espurna web interface using broad access point ?
(as showed in this guide - https://tinkerman.cat/magic-home-led-controller-espurnad/#lightbox-gallery-bzb7oR2s/1/ )

xoseperez, 2018-07-17:
Thank you @ColinShorts for the help. 
I'm closing this, @tomimarcincin: feel free to reopen if necessary.

closed 2018-07-17 by xoseperez