xoseperez/espurna #1057: how to make Sonoff S26 work with ESpurna?

akosmaroy, 2018-07-16:
Hi,

I got a bunch of Sonoff S26 plugs: https://www.itead.cc/sonoff-s26-wifi-smart-plug.html
while other Sonoff devices are supported by ESpurna, this is not. I wonder what it takes to make them work.
here are some images of the interior of the device: https://photos.app.goo.gl/ougSjDYB72tkxeCB9
OTA flashing of the ESpurna firmware doesn't seem to work: https://github.com/xoseperez/espurna/issues/1037
I'm new to ESpurna in general, thus any pointers, ideas welcome

Akos

battika, 2018-07-16:
Szia Ákos!

I have mine in transit can't wait getting it. If I manage to find out anything I will let you know. I will try to flash the S20 image like you did in #1037

xoseperez, 2018-07-16:
The S20 image should work with the S26. OTA flashing with the stock firmware might not work, I'm unsure about that, but you should be able to flash it wired using the pads in the WiFi module and the button to enter flash mode.

https://www.cnx-software.com/wp-content/uploads/2018/06/ESP8266-Sonoff-S26-Large.jpg

AlbertWeterings, 2018-07-16:
Wow nice design these S26

closed 2018-08-04 by xoseperez