xoseperez/espurna #1062: New Sync mode special for motors

glarfs, 2018-07-19:
Hi,
I'm using a 4ch board with espurna firmware, activating cover motors. The configuration I use:

1- motor A up
2- motor A down
3- motor B up
4- motor B down

At hardware level is configured to be only one motor up or down ( by default up will be used if both 1 and 2 relays are active (11) ). So on relay1=1 and relay2=1 will cause motor A up.

I checked the code as I wanted to add a new sync method "Zero or One switches  active on Pair relays" that allows only one on every 2 relays on but is not as clear/clean as the current options.

The problem is that If I use Zero or One  switches active I can't open both motors up at the same time.
I've also tried group sync but I couldn't find a good way to do it.

I also have another board with a single motor (relay1 and relay2) and a light on the relay3 and the relay4. So this new functionality won't apply on this other case, but that can be addressed on another issue.

I've also though to use a mask or something on each relay to specify the group.

AlbertWeterings, 2018-07-19:
Hi glarfs,

Maybe something to consider, I use one relay to set the motor direction 'nc' (off) is up the second relay is starting the motor. This way it is physical not possible to have UP and DOWN activated at the same time. I also found the self-induction of the motor can cause a reasonable spark inside a normal relay so I replaced the relay that activate the motor with a solid state version. The direction relay is a normal one as it will never switch when the motor is running it is not stressed.

hyteoo, 2018-07-20:
@glarfs from an electrical engineering standpoint, I would recommend you use a H-bridge controller which would have several benefits over relays, like:
- quiet
- control speed with PWM
- integrated protection against ShortCircuit, Overload, etc
- cheaper
- practically no way of accidentally driving a motor up&down at the same time

I could give an example that would fit your dual motor setup: http://www.ti.com/product/DRV8848
Of course everything depends on the motor power so if you could give us more details, that would be great ;)

xoseperez, 2018-07-26:
I think hardware-based control is a must when using motors, but you said you already have that covered.
From the software point of view, I like the relay groups idea. By default, all relays will belong to the same group and it will be the current behavior, but you could specify groups of 2 (or 2, 4) relays together. Then the sync options will apply per-group.

biot, 2018-10-08:
Just adding my vote to this. I have roller shutters I need to control with a Sonoff Dual R2, in a similar configuration: each relay is connected to a direction of the motor, and the Sonoff set to "Zero or one switches active" mode. However I want to use a single button to control the whole thing, where the button would just toggle the shutters opening or closing. Relay groups is the only thing missing for me here.

Ideally there would also be a feature where pressing the button while either relay is closed, would open it -- stopping a running motor. Next press would then close the other relay in the group. In other words, exactly like a single-button garage door opener.