xoseperez/espurna #1063: Support for MHZ19

jsponz, 2018-07-20:
When compiling for adding support for MHZ19 to the board NodeMCU Lolin, I get this error:

In file included from sketch/config/sensors.h:757:0,
                 from sketch/config/all.h:33,
                 from /Users/jorgesanchez/git/espurna/code/espurna/espurna.ino:22:
sketch/config/../sensors/MHZ19Sensor.h: In member function 'virtual void MHZ19Sensor::begin()':
sketch/config/../sensors/MHZ19Sensor.h:79:22: error: 'class SoftwareSerial' has no member named 'enableIntTx'
             _serial->enableIntTx(false);
                      ^
Multiple libraries were found for "SoftwareSerial.h"
 Used: /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/SoftwareSerial
 Not used: /Users/jorgesanchez/Documents/Arduino/libraries/espsoftwareserial-master
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Hardware file:

        // MHZ19
        #define MHZ19_SUPPORT       1
        #define MHZ19_RX_PIN        5
        #define MHZ19_TX_PIN        8

Any idea?

Valcob, 2018-07-20:
@jsponz, looks like you are using a wrong library, please always check you libs first ;) and the message is clearly saying that:
 **Used:** /Users/jorgesanchez/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/SoftwareSerial
**Not used:** /Users/jorgesanchez/Documents/Arduino/libraries/espsoftwareserial-master

If you check the platformio.ini somewhere around line 76 you'll find the lib  https://github.com/plerup/espsoftwareserial#3.4.1  intall it cause 
https://github.com/plerup/espsoftwareserial/blob/master/SoftwareSerial.h  has the missing method line 45 ;)
Hope it will help you, have fun and happy coding )

jsponz, 2018-07-20:
@Valcob Thanks for your answer. 

I am using Arduido IDE. Where should I make the change?

Valcob, 2018-07-20:
@jsponz, https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment
In short you need to go in both places and remove that folders  [.../SoftwareSerial] and [.../espsoftwareserial-master] respectively then in the last folder [/Users/jorgesanchez/Documents/Arduino/libraries/]  just issue the git clone command with the URL from the previous replay that should do the trick.

Also please check these: 
https://github.com/xoseperez/espurna/wiki/ArduinoIDE#installing-dependencies
https://github.com/xoseperez/espurna/wiki/ArduinoIDE#installing-libraries-manually   <<-- this one should guide you on how to install the required libs that are note available in lib manager

jsponz, 2018-07-20:
@Valcob Thanks for the hints. 

I have followed all of them and now the situation is as follows:

Build options changed, rebuilding all
In file included from sketch/config/sensors.h:757:0,
                 from sketch/config/all.h:33,
                 from /Users/jorgesanchez/git/espurna/code/espurna/espurna.ino:22:
sketch/config/../sensors/MHZ19Sensor.h: In member function 'virtual void MHZ19Sensor::begin()':
sketch/config/../sensors/MHZ19Sensor.h:79:22: error: 'class SoftwareSerial' has no member named 'enableIntTx'
             _serial->enableIntTx(false);
                      ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Any further help? Thanks in advance.

Valcob, 2018-07-20:
@jsponz 
Nothing changed )) (other than 2 folders were removed)
1. Check the src folder of the cloned lib. There you'll find the SoftwareSerial.h file check it for enableIntTx occurrence some where around line 45 if there is nothing then repeat the step with (git clone the master of that lib)/(manually install it from the zip downloaded from the same git repo)   
2. You can try restarting Arduino IDE to exclude libs being cached

jsponz, 2018-07-20:
Some findings:

Recommended ESPSoftwareSerial library (https://github.com/krosk93/espsoftwareserial) does not have 'enableIntTx'...

If I change to https://github.com/plerup/espsoftwareserial

It compiles, but I don't see the sensor on the web panel...

Valcob, 2018-07-20:
Well congrats, finally you have the right lib. 
And now it's totally another story. This will involve some debugging using the serial monitor window and not only.  
Are there any error messages like failed to init and so on...

jsponz, 2018-07-20:
Info on the website should be updated :-) 

Anyway, I get it finally on the panel, but:

[188610] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
[194617] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
[200621] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
[206624] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
[212633] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
[218641] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
[224649] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
[230655] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
[236659] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
[242669] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)

I have changed the pins, but with the same result. 

Any idea?

Valcob, 2018-07-20:
//------------------------------------------------------------------------------
// MHZ19 CO2 sensor
// Enable support by passing MHZ19_SUPPORT=1 build flag
//------------------------------------------------------------------------------
.....
#ifndef MHZ19_RX_PIN
#define MHZ19_RX_PIN                    13
#endif

#ifndef MHZ19_TX_PIN
#define MHZ19_TX_PIN                    15
#endif
.... 
In config/sensor.h the pins are defined as 13 and 15. Are you listening on the right pins?

jsponz, 2018-07-20:
I use pins 14 and 15. They are connected to the MHZ19. 

I have defined them in /config/hardware.h under my board:

        // MHZ19
        #define MHZ19_SUPPORT       1
        #define MHZ19_RX_PIN        15
        #define MHZ19_TX_PIN        14

Valcob, 2018-07-20:
Looks like you need to swap the pins )) from the line of code[... SwSerial(%u,%u)", _pin_rx, _pin_tx ...] we are printing the rx pin number first then tx pin number and from your error 
[242669] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
RX pin is 14 and TX is 15  and your config says
 #define MHZ19_RX_PIN        15
 #define MHZ19_TX_PIN        14

Check the device connection and pin configuration

Valcob, 2018-07-20:
https://www.winsen-sensor.com/d/files/PDF/Infrared%20Gas%20Sensor/NDIR%20CO2%20SENSOR/MH-Z19%20CO2%20Ver1.0.pdf
Please ensure you are using the right things to connect the sensor
Also please not that sensor uses 3.6 - 5V  so it might not work properly if connected to 3.3 line 
From the manual
**7.2 Transmit Data**
Vin-5V power
GND- Power Ground
RXD connect sensor TXD
TXD connect sensor RXD
..... 

so the sensor TX pin 3 should be connected to the Lolin's  RX  15 in your case and sensor RX pin 2 to the Lolin's TX 14 in your case

jsponz, 2018-07-20:
@Valcob Thanks!

Just from the hardware:

MHZ19
Vin = 4.54 V
GND -> GND
RX (pin 2) -> Pin 15
TX (pin 3) -> Pin 14

I have configured the hardware file accordingly, but nothing...

Valcob, 2018-07-20:
From your definitions you have RX on LOLIN set to pin 15 
#define MHZ19_RX_PIN 15
#define MHZ19_TX_PIN 14
this way wont work... 
You need connection like  RX(2) <---> TX(14) and TX(3) <---> RX(15) respectively

jsponz, 2018-07-20:
Yes, I know. But I have changed the definitions... without success...

Valcob, 2018-07-20:
Sorry, I don't have an MHZ19 right now so I can't check the code is working for myself but I hope someone who has this sensor can help you or fix the code if something was broken...

jsponz, 2018-07-20:
@Valcob Thanks for your help!

I guess that there is something broken...

AlbertWeterings, 2018-07-20:
In sensors.h no baudrate is set for this sensor. Sensor baudrate is 9600. I can't check on my tablet (no search in files option) but I think baudrate is set by default at 19k2. Hope this helps

jsponz, 2018-07-21:
it is defined in:

https://github.com/xoseperez/espurna/blob/dev/code/espurna/sensors/MHZ19Sensor.h

line 80.

jsponz, 2018-07-23:
Any further support?

Valcob, 2018-07-23:
@jsponz I would like to help you but I don't have this sensor now.... though we can try one thing just to see what the result buffer contains
add these lines in line 185 of https://github.com/xoseperez/espurna/blob/dev/code/espurna/sensors/MHZ19Sensor.h
right after line 184 with _write(MHZ19_GETPPM, buffer);
Serial.println(buffer[0]);
Serial.println(buffer[1]);
Serial.println(buffer[2]);
Serial.println(buffer[3]);
Serial.println(buffer[4]);
Serial.println(buffer[5]);
Serial.println(buffer[6]);
Serial.println(buffer[7]);
Serial.println(buffer[8]);

then just post the results from log window in here

jsponz, 2018-07-24:
@Valcob Thanks for your hint. I will test it on Thursday and I will get back to you.

xoseperez, 2018-07-26:
I have an MHZ19 I can test but I won't be able to do it until mid-August since I'm on holidays.
I have changed the recommended library for the SoftwareSerial in the wiki. Thank you.

jsponz, 2018-07-27:
@Valcob, please find below the results:

0
0
0
0
0
0
0
0
0
[031079] [SENSOR] Error reading data from MHZ19 @ SwSerial(15,14) (error: 5)
0
0
0
0
0
0
0
0
0
[037080] [SENSOR] Error reading data from MHZ19 @ SwSerial(15,14) (error: 5)
0
0
0
0
0
0
0
0
0
[043084] [SENSOR] Error reading data from MHZ19 @ SwSerial(15,14) (error: 5)

and if I switch the serial port:

0
0
0
0
0
0
0
0
0
[066483] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
0
0
0
0
0
0
0
0
0
[072491] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)
0
0
0
0
0
0
0
0
0
[078498] [SENSOR] Error reading data from MHZ19 @ SwSerial(14,15) (error: 5)

Any help? :-) 

Thanks!

jsponz, 2018-07-27:
@Valcob, it works!

I have connected Rx/Tx to GPIO 4 and 5 and it works!

Now it returns:

255 
134 
1 
139 
76 
0
0
0
162

What are other pins that I could use? IO4 and IO5 are being used for I2C. 

Thanks!

jsponz, 2018-07-27:
Final pinout:

- IO12 & IO16 -> MHZ19
- IO13 -> Relay
- IO2 -> DS18B20 
- IO4 & IO5 -> I2C
- IO0 -> DHT11 (yes, it is also a temperature sensor)

Everything possible :) 

Thanks!

Valcob, 2018-07-27:
You are very welcome!  now you can close the issue ticket 

Thanks and enjoy

jsponz, 2018-07-27:
Sure!

closed 2018-07-27 by jsponz