xoseperez/espurna #1064: webserver problem after a while

ServerCare, 2018-07-20:
I have found intermitted issues on some of my POWR2's, three of the 5 devices I currently am using are having the same symptoms: when I try to access the web -interface It does an authentication challenge, but after providing my credentials, rather than showing me the web interface, it starts a download of a file. The file is called ' download.crdownload' and it never seems to stop (and seems empty)

Interestingly the REST API (which is being used to monitor the status every minute) seems to work fine...

When I try to logon on telnet things seem to work well, not sure what to check next.

I can reset the device through telnet, but unfortunatly when you do that the relay switches off and on again (can this be fixed? should I open an issue for that?), effectively rebooting my servers that are being monitored that way ...

xoseperez, 2018-07-26:
I think the problem with the web interface could be memory related. I would check the free heap at boot and monitor the value (maybe logging that value in the MQTT heartbeat messages) to see if there is a memory leak somewhere. In your log, it shows 11560bytes and that a bit low. Under 10k memory issues are possible when doing stuff like web.

As for the boot issue. When the controller resets it sets all GPIOs low.

ServerCare, 2018-07-26:
on one of the powr2's currently experiencing the issue:

[385250] [WEBSERVER] Request: GET /api/power
[385252] [API] Sending response '130'
heap

[391849] Free HEAP: 7760 bytes
[391850] +OK
[399072] [WEBSERVER] Request: GET /index.html
heap

[402481] Free HEAP: 4480 bytes
[402482] +OK
info

[413343]

[413344] [INIT] ESPURNA 1.13.0
[413345] [INIT] xose.perez@gmail.com
[413345] [INIT] http://tinkerman.cat

[413345] [INIT] CPU chip ID: 0x4AD078
[413345] [INIT] CPU frequency: 80 MHz
[413345] [INIT] SDK version: 1.5.3(aec24ac9)
[413346] [INIT] Core version: 2.3.0
[413346] [INIT] Core revision:
[413346]
[413346] [INIT] Flash chip ID: 0x1640EF
[413347] [INIT] Flash speed: 40000000 Hz
[413347] [INIT] Flash mode: DOUT
[413348]
[413348] [INIT] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[413349] [INIT] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[413349] [INIT] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[413349] [INIT] Firmware size       :   477472 bytes /  117 sectors (   1 to  117)
[413350] [INIT] Max OTA size        :   544768 bytes /  133 sectors ( 118 to  250)
[413350] [INIT] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[413350] [INIT] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[413350]
[413351] [INIT] EEPROM sectors: 251, 250
[413351]
[413352] [INIT] BOARD: ITEAD_SONOFF_POW_R2
[413352] [INIT] SUPPORT: ALEXA BROKER DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB
[413352] [INIT] SENSORS: CSE7766
[413352]
[413353] [INIT] Last reset reason: Power on

xoseperez, 2018-07-26:
It is obviously a memory problem. It won't work with as little as 4480bytes of heap...
Question is where is the memory going? I have a POW2 monitoring my washer machine and it works without problem, heap reporting ~19Kb.

ServerCare, 2018-07-26:
I noticed a lot of my devices were pointing to a no longer existing DNS (also noticed NTP issues) so after fixing that I upgraded them all to 1.13.1.

I'll monitor the progress...

ServerCare, 2018-08-02:
As promised... I monitored the progress...

AT this point I am confident the issue was introduced by a misconfiguration of the DNS settings in the witreless settings. For all nonb-responsive devices they were pointing to an 'old' IP for a DNS that was removed. Once a proper DNS was entered; problems over....

closed 2018-08-02 by ServerCare