xoseperez/espurna #1069: Problem integrating sonoff basic with alexa

popoff1998, 2018-07-23:
I used esp easy for my sonoffs to integrate with domoticz. Now I want to try it with espurna to use the Alexa integration feature.

I see that the wemo emulation doesn't supports the wemo native emulation, so I can't check in the wemo app that the sonoff is emulating a legit wemo device.

When open the alexa app, in home automation, I can't discover the sonoff with espurna. When I install the wemo skill on my echo device, the skill ask me for the mac address of one wemo device on my network and the wifi ssid. I use the sonoff mac address but the skill instalation fails.

Is needed ti have at least one legit wemo device added previously to use the espurna alexa integration?

I can't find many information about the alexa integration of a sonoff device with espurna.

Regards.

xoseperez, 2018-07-26:
You have to have the Alexa support enabled in the web interface of ESPurna and the do the discovery. If it does not work I'll need to know what kind of Alexa device you have. Are you using a prebuild binary or are you building it yourself?

loglux, 2018-07-29:
It doesn't work for me. I have Amazon Echo Show

popoff1998, 2018-07-30:
No way.

Of course I have "Alexa support enabled" in espurna web interface. The device is a sonoff basic that is connected rigth to my wifi. My alexa device is amazon echo. All other switches that work with alexa are ok. I have other sonoff basic and sonoff touch with original ewelink firmware, two sockets with that use eFamilyCloud, even a sonoff basic with esp easy integrated in domoticz that works with alexa using controllicz.

I tried with prebuilt binary and with an own build downloading the last git version and building espurna with platformio. The flash is ok and I have the espurna web interface ok.

I think that the problem, as I said, is with the wemo skill for alexa. Espurna sonoff basic isn't detected by the wemo app, so I can't add it to the wemo cloud. When I try to activate the wemo skill at amazon.co.uk (this skill isn't yet in amazon.es that is my default store, but for instance efamilycloud and ewelink skills aren't and I installed from the uk store and are working ok); the wemo skill detects that my account haven't any wemo device registered (without an original wemo device or and espurna that works with te wemo app I can't register). So the wemo skill open a dialog asking for the name of my wifi and the mac address of the wemo device. I put both but the skill installation fails.

xoseperez, 2018-08-04:
The Alexa support in ESPurna is based on faking a wemo switch but the API is not 100% compatible since the specification is closed and based on a reverse engineer work. This means it won't show up in the Alexa web or app, and it doesn't work with the Wemo Skill.

I'd suggest you to check the examples in the [FauxmoESP](https://bitbucket.org/xoseperez/fauxmoesp/src/master/) project to see if the devices is supported and then go back to ESPurna.

closed 2018-10-10 by stale[bot]