xoseperez/espurna #1070: Livolo support for RF-Bridge?

olili, 2018-07-23:
Does anybody see a chance to get Livolo protocol support included in RF-Bridge?
Details of the protocoll see here:
https://forum.arduino.cc/index.php?topic=153525.180

Thx in advance
O

xoseperez, 2018-07-26:
The protocol seems very simple and quite similar to the standard the RFBridge uses but it lacks the header bytes. The default protocol for the RF Bridge uses is 9 bytes (2 for the sync length, 2 for the low, 2 for the high and 3 for the code itself). In the Linovo protocol the first 6 bytes are assumed to be fixed lengths and only transmits the 3-byte code.

Anyway, you can use it with the RF Bridge using the RFB_DIRECT hack or the RF_RAW_SUPPORT hack. The first one requires modifying the board (https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack) and the second one requires flashing the secondary microcontroller in the RF Bridge with a custom firmware.

olili, 2018-07-26:
Dear  xoseperez ,

thx. 
I already installed the custom firmware on the the secondary micro.
But until now I was not able to get any Livolo command to be received and forwarded via MQTT. 

Any hint?  

O.

closed 2018-10-01 by stale[bot]