xoseperez/espurna #1071: Dimming

jsponz, 2018-07-24:
Hi,

I want to use PWM to dimmer. 

I have bought a device such as this: https://www.aliexpress.com/item/AC-Light-Dimmer-Module-for-PWM-control-1-Channel-3-3V-5V-logic-AC-50-60hz/32802025086.html

How should I use with ESPURNA? 

I guess that it is similar to the LED controllers with dimming. 

Just let me know your thoughts!

Thanks!

ColinShorts, 2018-07-24:
If you are comfortable working with the voltages involved, the principal seems to be fairly simple. I don't think that Espurna has the code to handle it right now, so you'd need to find a suitable library to handle the dimming for this module as it has a pin that is used to register when the AC power crosses 0V, and another pin for pwm. Essentially the zero crossing determines the timing of where the voltage is cut by the pwm. Of course, that's just my limited understanding, and I may be entirely wrong, but in any case I wouldn't personally recommend this to anyone that wasn't trained/comfortable with the risks involved.

In case you hadn't realised, you would also need an esp8266 board to interface, from which you'd need at least two free gpio pins (pretty much any standard esp should be usable). The dimmer module itself looks to be capable of 3.3V on the "safe" side, so at least in theory it won't need very much in the way of external components.

Don't electrocute yourself!

ColinShorts, 2018-07-24:
Actually... There might be a bit more to it than I thought, but this instructable has more detail and even mentions your dimmer module:
https://www.instructables.com/id/Arduino-controlled-light-dimmer-The-circuit/?amp_page=true

xoseperez, 2018-07-26:
Yes, examples are a bit hard to understand but it doesn't look too complicated to implement. Current ESPurna dimmers use PWM whilst this one uses PSM (Pulse Skip Modulation). I guess it requires a nw LIGHT_PROVIDER.

lblabr, 2020-04-14:
are there any news to that ? i would also be interested ... i think, there is one thing you need one pin to get synchronzied to the zero cross. i also thought already to use the existing pwm module ... but the frequency is think the best way is to synchronzize the pwm from external with zero cross....after that you might have a pwm freqency about 100Hz where you may use the existing pwm implementation .... 

what do you think how complicate is that to implement ?