xoseperez/espurna #1072: webinterface/switches: customizable labels and general understanding - newbeee

lblabr, 2018-07-24:
i´m quite new to your project and it feels great! Really good job.

i´d like to use espurna on different ESP´s to control the water for my garden, sprinkling and pumps.
i have some (little) knowledge about programming and html, css and that stuff.

two things i would like to do:

1) switching one relay in specified interval, each 10 minutes on for 5 minutes, 5 minutes off .... 
2) i would like to have speakinng labels on the switches

could you give me some hints where and how to start ?

Idea for 1)
 i think i will just add an attribut to the scheduler data, interval, and after executing the execution time of the scheduling will be updated with an new one  

Ideas for 2)
great would by a configurable Label for the Relay, like mode or gpio, better may be through Webinterface

the Relaylist ist generated could give me some hints where to start ? what would fit in the concept of espurna ?

I´m uisng Nodemcu with SPIFFS, could i also load the web stuff on spfiffs to avoid the build process ?

Thanks in advance

Lars

xoseperez, 2018-07-26:
The first option could be easily developed as a new pulse mode. Now there are "no pulse", "normally on" and "normally off". A "toggling" option would use the defined interval to toggle on/off endlessly.

The second option is already requested

lblabr, 2018-07-27:
Hi Xose,

good idea! to have an adjustable on/off ratio, i would set the intervall to lets say 10 minutes (toggle) and set the pulse duration by the setting in the realay configuration. so the toggle should take the actual status as basis for the toggeling. in that case the following should work: 

interval: 10 minutes
Pulseduration 1 minute
switches on in schedule, switches off after 60 seconds, "toggles" on after another 9 minutes...

interval: 10 minutes
Pulseduration 6 minutes:
switches on in schedule, switches off after 6 minutes, "toggles" on after another 4 minutes...

interval: 10 minutes
Pulseduration 0 minutes:
switches on schedule , "toggles" off after another 10 minutes...

just to have it clear

thanks in advance

Lars

skyynet, 2019-04-07:
I also would like to label the switches in web interface. Would be pretty easy in web interface tab "switches".

closed 2024-08-22 by mcspr