xoseperez/espurna #1073: Relay turns on at boot.

jchrzaniuk, 2018-07-24:
Hi,
I've noticed, that in this configuration (ITEAD_SONOFF_BASIC):
```
#define BUTTON1_PIN	0
#define BUTTON1_MODE	BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
#define BUTTON1_RELAY	1
#define BUTTON2_PIN	14
#define BUTTON2_MODE	BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
#define BUTTON2_RELAY	1
```

if BUTTON2 is closed (BUTTON2_PIN is LOW) and Espurna reboots, then RELAY1 always turns on (even if the state before reset was off).

BauerPh, 2018-07-24:
Try to change BUTTON_DEFAULT_HIGH to BUTTON_DEFAULT_LOW
Compile and upload

BauerPh, 2018-07-24:
I have the same issue with BUTTON_MODE = BUTTON_SWITCH.
I think the button state should be read during setup and should not fire a Click event?

closed 2018-09-02 by xoseperez