xoseperez/espurna #1074: Single Sonoff Basic showing as 2 switches in Home Assistant

MaxiStarling56, 2018-07-25:
Hi,
if you read the title you know the problem, i used the code that Espurna gave me (home assistant discover, configuration code) but in the home assistant overview he show like if there were 2 lights (watch the immages).
I hope someon can help me and thank for the attention.

[image]
[image]
[image]
[image]

mcspr, 2018-07-25:
Configuration gets doubled :) Exact same config of `switch:` is also received by the mqtt discovery (`mqtt.discovery: true`). You might want to keep configuration.yaml entry and disable **Discover** on espurna side.

xoseperez, 2018-07-26:
Yes, I would first check if you have Discovery mode enabled. The device is probably sending the configuration to HA and it gets doubled.

Carmelo66, 2018-07-28:
It seems to me that a switch works and the other only reads the status if turned on or off

closed 2018-10-03 by stale[bot]