xoseperez/espurna #1080: Incese number of switches on sonoff rf bridge

adas4190, 2018-07-26:
Hi. Is it possible to increse number of switches? If so how can I do it pls? I'm no programer 😞

hyteoo, 2018-07-27:
If you go to code/espurna/config/hardware.h, [here](https://github.com/xoseperez/espurna/blob/be7f535d149359763e62e2bd7f5943c31941270d/code/espurna/config/hardware.h#L665) is the definitionof the RF Bridge

You can add extra buttons in the "// Buttons" section by following the existing example or other devices' example.

More info on that also in the [wiki](https://github.com/xoseperez/espurna/wiki/Buttons-and-switches)
Just make sure you use unused GPIO pins for that.

xoseperez, 2018-08-04:
Just change the DUMMY_RELAY_COUNT value in the hardware.h file

https://github.com/xoseperez/espurna/blob/be7f535d149359763e62e2bd7f5943c31941270d/code/espurna/config/hardware.h#L674

closed 2018-10-10 by stale[bot]