xoseperez/espurna #1088: two forced reboot are needed after the initial flash to get to the web interface

wsw70, 2018-07-29:
I just flashed (todays's dev, commit be7f535) a new Sonoff RF Bridge and following the [Wiki details on first boot](https://github.com/xoseperez/espurna/wiki/Configuration) I could connect to the AP.

When trying to get to http://192.168.4.1, I got a blank page. **I had to force reboot** (power off/on).

Going then to http://192.168.4.1, I got the initial password change page, but clicking 'Update' seemingly did not do anything (no change on the page).  **I had to force reboot again.**

Subsequent connections to the interface were fine, including after reboots.

It was some time I did not flash a device but I seem to remember that I had similar issues before as well (but did not document/realize clearly).

BauerPh, 2018-07-30:
You may have to do a manual reboot after flashing if your device/glasher does not do it automatically.
After pressing Update on the initial update page you also change the Wifi AP Password. So you need to reconnect to your device with the new password.
The behaviour you described seems "okay" to me.

xoseperez, 2018-08-04:
After flashing the device via wire a manual reboot is required as @BauerPh states. This is an issue with the ESP8266 SDK, not ESPurna related. The second reboot shouldn't be necessary, it could be some kind of caching issue.

BauerPh, 2018-08-04:
Agreed,
on windows 10  you have to change the wifi password to the new one. You dont have to reboot the device.

closed 2018-10-10 by stale[bot]