xoseperez/espurna #1089: Crash and reboot when measuring and reporting every second

atoomkern, 2018-07-30:
I am using the BlitzWolf BWSHP2 in combination with Espurna to measure and report the energy use every second. Unfortunatly the device randomly crashes and reboots after a few minutes up to ten hours. 

I tried Thingspeak and MQTT, changing the version of espurna from 1.13.1 to the current dev branch and I tried all the three Arduino Core versions.

Any suggestions what could be causing these crashes?

atoomkern, 2018-07-31:
Below the crash log. Any help is appreciated.

[193247] [WEBSOCKET] Requested action: dbgcmd
[193252] [DEBUG] Latest crash was at 66610044 ms after boot
[193254] [DEBUG] Reason of restart: 3
[193255] [DEBUG] Exception cause: 4
[193256] [DEBUG] epc1=0x402016d4 epc2=0x00000000 epc3=0x00000000
[193259] [DEBUG] excvaddr=0x00000000 depc=0x00000000
[193264] [DEBUG] >>>stack>>>
[DEBUG] 3ffffe80: 00000000 00000000 00000000 00000000 
[DEBUG] 3ffffe90: 401075f8 00000000 3fff79cc 40226f0d 
[DEBUG] 3ffffea0: 0000002f 03f8597e 3fff0dcc 40223bed 
[DEBUG] 3ffffeb0: 401075f8 00000000 00000000 ffffffff 
[DEBUG] 3ffffec0: 00000000 3fff6604 00000000 3fff6f64 
[DEBUG] 3ffffed0: 3fff94a0 00000007 3fff6f94 3fff94a0 
[DEBUG] 3ffffee0: 401075f8 00000000 3fff6604

xoseperez, 2018-08-04:
The exception seems to be related to an interrupt. Asking for the consumption every second seems a little overkill. To debug it further you should decode the exception stack, check [this](https://github.com/me-no-dev/EspExceptionDecoder) to do it.

closed 2018-10-10 by stale[bot]