xoseperez/espurna #1090: RFBridge send the learned RF code 4 timews in stead of ones.

Frans1960, 2018-07-30:
I flashed the ESP and the RF chip with firmware and teaches the RF codes of KUKA (KlikAanKlikUit) remote control.
I configured blinds in Domoticz and when the RfBridge recieves the switch command it sends the RF codes 3 of 4 times instead of one.
Can I configure this times sending some were on the WEB Gui?

This is the logging from the RFBridge:
[496908] [WEBSOCKET] Requested action: rfblearn
[496910] [RFBRIDGE] Sending LEARN
[496987] [RFBRIDGE] Action 0xA0
[501344] [RFBRIDGE] Action 0xA3
[501348] [RFBRIDGE] Sending ACK
[501352] [RFBRIDGE] Received message '2C56010403F2440014'
[501353] [RFBRIDGE] Learn success
[501354] [RFBRIDGE] Storing 0-ON => '2C56010403F2440014'
[506221] [WEBSOCKET] Requested action: rfbsend
[506223] [RFBRIDGE] Storing 0-ON => '2C56010403F2440014'
[510365] [WEBSOCKET] Requested action: rfblearn
[510367] [RFBRIDGE] Sending LEARN
[510444] [RFBRIDGE] Action 0xA0
[514450] [RFBRIDGE] Action 0xA3
[514454] [RFBRIDGE] Sending ACK
[514458] [RFBRIDGE] Received message '2C74010E03F2440015'
[514459] [RFBRIDGE] Learn success
[514460] [RFBRIDGE] Storing 0-OFF => '2C74010E03F2440015'
[519678] [WEBSOCKET] Requested action: rfbsend
[519680] [RFBRIDGE] Storing 0-OFF => '2C74010E03F2440015'
[521642] [WEBSOCKET] Parsing configuration data
[605224] [MQTT] Sending RfBridge/app => ESPURNA (PID 1)
[615846] [MQTT] Received domoticz/out => {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014058",
   "idx" : 8,
   "name" : "VertikaalScreen",
   "nvalue" : 0,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "Blinds",
   "unit" : 1
}

[615856] [DOMOTICZ] Received value 0 for IDX 8
[615859] [RFBRIDGE] Enqueuing MESSAGE '2C74010E03F2440015' 4 time(s)
[616734] [RFBRIDGE] Action 0xA0
[617233] [RFBRIDGE] Action 0xA0
[617734] [RFBRIDGE] Action 0xA0
[626335] [MQTT] Received domoticz/out => {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014058",
   "idx" : 8,
   "name" : "VertikaalScreen",
   "nvalue" : 0,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "Blinds",
   "unit" : 1
}

[626344] [DOMOTICZ] Received value 0 for IDX 8
[626347] [RFBRIDGE] Enqueuing MESSAGE '2C74010E03F2440015' 4 time(s)
[627222] [RFBRIDGE] Action 0xA0
[628221] [RFBRIDGE] Action 0xA0
[628722] [RFBRIDGE] Action 0xA0
[632235] [MQTT] Received domoticz/out => {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014058",
   "idx" : 8,
   "name" : "VertikaalScreen",
   "nvalue" : 1,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "Blinds",
   "unit" : 1
}

[632244] [DOMOTICZ] Received value 1 for IDX 8
[632246] [RELAY] #0 scheduled ON in 0 ms
[632247] [RELAY] #0 set to ON
[632250] [RFBRIDGE] Enqueuing MESSAGE '2C56010403F2440014' 4 time(s)
[632253] [MQTT] Sending RfBridge/relay/0 => 1 (PID 1)
[632445] [MQTT] Received domoticz/out => {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014058",
   "idx" : 8,
   "name" : "VertikaalScreen",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "Blinds",
   "unit" : 1
}

xoseperez, 2018-08-04:
When the ON and OFF codes for the same switch are different ESPurna sends the message 4 times by default. This is the expected behavior. If you want to change it you will have to edit the general.h file and change the RF_SEND_TIMES setting, rebuild and flash.

https://github.com/xoseperez/espurna/blob/f18c901c3f0f99211175c08a5b204311609f4447/code/espurna/config/general.h#L1026

Frans1960, 2018-08-10:
Okay, thank you for the info, it helps. Now it's sending not 4 times bud one's and than again, because it receives also and domoticz/in message for the RfBridge. (on state change of the RF channel/switch)
I use the RFBridge for sending and receiving RF Signals and I would like to configure or program each RF channel/switch  on the direction (receiving or sending RF code), so that the RfBridge only the receiving RF channel/switch sends a domoticz/in to the MQTT broker. 
Can you help me? Thanks in advance

Frans1960, 2018-08-13:
Yes that will do. With that I can select what switches sends messages to domoticz and what switches don't

xoseperez, 2018-08-14:
Then just set the IDX to 0 for those switches you don't want to send notifications...

Frans1960, 2018-08-14:
That I could do, but than I can't control them with Domoticz and that was my goal.

Frans1960, 2018-08-14:
I'm now using PlatformIO to build the firmware and use "poi run - itead-sonoff-rfbridge-direct" to build the firmware and it says success for the environment, but it seems that the direct version is not build. Can you help?

Frans1960, 2018-08-14:
Sorry, I don't use the direct, but with the altered firmware of the RFchip

Frans1960, 2018-08-14:
I solved the RF_RAW item

closed 2018-10-20 by stale[bot]