xoseperez/espurna #1091: RF Bridge does not learn any code

olli-r, 2018-07-31:
Hello,

maybe someone who could help me out:

I modified the RF bridge according to the documentation of the direct hack. I switched back the tiny switch on the circuit board, thereafter I installed espurna-1.13.1-itead-sonoff-rfbridge-direct.bin and logged in.
When trying to learn new RF codes, the beeper stays silent (expected behaviour...), but the red RF led stays off as well (is that supposed to be?).

However, I can't learn to a remote, tried two different rc models. When setting the "rfb-dump" to 2, no data is shown - no matter if in learn mode or not. Randomly, there is some data recieved, but it's there even if I don't press any button on the remote - so this may be another sender.

When measuring the two GPIO pins with an oscilliscope, there is a steady signal on the RX side. When a rc button is pressed, the signal significantly changes, so I expect the RX module to work correctly.

Does anyone have a hint what I am doing wrong?

Thanks,

Olli

xoseperez, 2018-08-04:
It looks like a connection problem but I can't say... maybe @wildwiz has an idea.

wildwiz, 2018-08-05:
Hi, the rfb-dump feature is not present on the official firmware, you need to use the one mentioned here https://github.com/xoseperez/espurna/pull/864 (it derives from 1.13.0, not from the latest version, but it is okay for testing). 
Based on your details, I would suggest first to check that you made the change appropriate to the type of board you have (pin 4 and 5 are swapped on some boards revisions). I would check that the demodulated RF noise (or transmission, when activating the remote) is present on the side of the resistor near to the ESP8266. You should have noise/signal with 3.3V swing on one pin, and nothing on the other. If you have something like 0.5V swing, then the pins are very likely swapped.
If you do indeed have signal on pin 4 of the ESP8266, then trying to upload the firmware from 864 might help to understand if it actually is a decodeable signal or not.

olli-r, 2018-08-11:
Hello, thanks for the replies! I already tested with the rf-dump supported firmware, I didn't write it clearly though - sorry.
I checked again with my oscilliscope, when I send data with espurna, there's a modulated signal going to the TX pin - so I assume that the pins are not swapped.
On the RX side there's always some kind of signal, looks more or less like background noise. When a button is pressed on the remote, there's a change in that signal.

Using the rf-dump feature, I recieve some data, but I has obviouly nothing to do with the button presses on my remote as it's recieved randomly every 20-30 seconds - even without pressing any button.

Could there be the possibility that my hardware is just broken?

Thanks for your help!

wildwiz, 2018-08-12:
Hi, the random noise outside transmissions is ok, as this kind of receivers pump up the sensitivity till they pick something - that's also why there are sync pulses before the actual transmission, to setup the receiver actual gain. receiving something every 20-30 seconds might be some other 433 device, like, for instance, maybe an external temperature sensor of a weather station. Assuming that your RX output isn't shorted, and your Xscope says it is not, and considering that you don't see signal during your remote transmissions, the logical conclusion would be that the receiver is not catching anything, whatever the reason. This is confirmed (always excluding the case of a short on the RX signal), by the red led not flashing. Even the uC output is not used, the input is still fed from RX and the led should blink more or less at the release of remote button. Maybe the remote uses another frequency ?

closed 2018-10-18 by stale[bot]