xoseperez/espurna #1096: Feature Request: Delay off on Sonoff 4Ch

wardaddyalpha, 2018-08-02:
Hi! I have an automatic curtain system which opens and closes when the corresponding line is pulled high for 10 seconds or so (which is the time it takes to open or close). I want to use the Sonoff 4ch template but want the 4 switches to turn the pin back low after the delay period. I want scheduling to work as well ideally in this same manner. I am not sure which sketch to modify if someone can point me in the right direction. Thank you!

flashy02, 2018-08-03:
Hi, this feature is already useable. You have only to log in your device and define under switches the pulse mode. For example if you want to automatically turn off after 5s, you have to define under Switch #X (GPIOX):
...
Pulse mode: Normally OFF
Pulse time (s): 5
...

wardaddyalpha, 2018-08-03:
Hi flashy02, thank you very much for the quick reply! I tried the pulse option but maybe I need to restart it. I will try again, thank you very much!
Another quick question, is there any option to turn these GPIO on or off directly with a weblink and without authentication like on Tasmota firmware? Thanks again!

flashy02, 2018-08-03:
I don't know all possibillities how to turn on/off a channel with espurna. Therefor you should read the wiki. For all devices I'm using in my HomeAutomation I use HW-Pushbuttons and/or MQTT with OpenHab.

xoseperez, 2018-10-05:
@wardaddyalpha you can use MQTT or the REST API to toggle a relay status. The REST API uses a key but no other authentication method.

closed 2018-10-05 by xoseperez