xoseperez/espurna #1097: xmw01esp & pl7211

gn0st1c, 2018-08-02:
inside the latest xiaomi enhanced power plug (with two usb ports), i see the following chips.

xmw01esp: esptool thinks it's a 8255 with 2mb flash. (the 25q16csig next to it is a  16mb flash)
pl7211: [2-Channel Power/Energy Metering Analog Front End](http://prolificusa.com/product/pl7211-2-channel-power-energy-metering-analog-front-end-afe/)

did anyone else has seen these before? 

----

[image]
[image]

xoseperez, 2018-08-04:
Wow, haven't seen these before. Do you have a link to the device page?

gn0st1c, 2018-08-05:
[Mi Smart Socket Enhanced Edition](http://item.mi.com/1180800005.html)

gn0st1c, 2018-08-13:
ok, found this post: [WHAT IS THE XMW01ESP IC?](https://www.esp8266.com/viewtopic.php?f=13&t=17294)

> I disassembled a xiaomi desk lamp and found an Espressif "XMW01ESP" inside.

----

it seems xiaomi is using esp chips now. maybe espressif is producing a special batch for mi devices.
i'll close this, as this is not an issue. i'll let you know when i know/test more.

closed 2018-08-13 by gn0st1c