xoseperez/espurna #1099: [discussion] Should we move MQTT topics to PROGMEM?

Valcob, 2018-08-03:
@xoseperez Do we have any reason to keep all MQTT topics in general.h or we can maybe move them to PROGMEM?

Valcob, 2018-08-13:
Well indeed there are tons of things that a strings and that are kept in RAM and by simply moving them to PROGMEM is not a solution cause we will end up with a bloated ROM with all kind of strings. E.g I am using only temperature and humidity so there is no need to keep the rest of topic constants even the rest of measurement units. I don't know maybe I am to concerned about the space being used in both RAM and ROM but we have to some how be more dynamic/specific with these constants and try to leave aside all that  we are not gonna use for a given build. Anyways I've already started a general rework/cleanup and I am not sure that these changes will be accepted as they will affect too many files. But I'll create some pull requests as soon as I finish these changes but first I have to finish some other cool things like a general controller (with a lot of bells and whistles) instead of scheduler and adding some OLED and EPD display support

xoseperez, 2018-08-13:
@Valcob that controller thing sounds great. 

About the strings, I have always had the feeling it would take too much time to move things around so I have always delayed this. If you come up with a solution it will be very welcome.

Valcob, 2018-08-25:
@xoseperez Hi, sorry to put this in here but don't know whether there is a discord or some other communication channel for general QA stuff but in the process of moving all strings to progmem found one particular thing regarding the app name and app version one is "app_name" and "app" and respectively "app_version" and "version" so the last shorter strings are used in /discovery GET call of the web.ino. My question is what is the purpose of this endpoint, does it follows some protocol can I send instead of app the app_name?

xoseperez, 2018-08-25:
The `/discovery` endpoint is not used at the moment. The idea is to be able to id all ESPurna devices from a web browser by just GETting this URL. In the future, this will allow having a common manager somewhere in the network.

Do you mean here?

https://github.com/xoseperez/espurna/blob/f840a5420f702475b02bbc0127e90450f8f13112/code/espurna/web.ino#L59-L75

Valcob, 2018-08-25:
Yep right there. So basically I want to merge 
root["app_name"]     = APP_NAME;   // from ws.ino 
and
root["app"]     = APP_NAME;   // from web.ino 

into one app_name. As I understand discovery is a custom functionality so I can change the string to app_name with no worries that I'll break something

xoseperez, 2018-08-26:
Yes, o problem in changing that.