xoseperez/espurna #1100: Generic 4 Ch ESP01S relay support

vaidyasr, 2018-08-04:
I am trying to use the Espurna for my DIY 4 Channel ESP01s relay and I can't control the switches via web interface. I am trying to follow the below guide for connection. https://arduinodiy.wordpress.com/2018/06/14/simple-wifi-relay-board-or-diy-sonoff-sv-sonoff-4-channel-2/
Do I need to change anything apart from adding a new '4CH definition' in the hardware.h ?

ColinShorts, 2018-08-04:
You could base the config on the Sonoff 4ch (not pro), remove the button definitions, update the gpio pin assignment to suit, and I think switch to RELAY_TYPE_INVERSE if the relay module you are using switches the relays on when driving the pin low.

If you are submitting a pull request then you'll probably want to update 4 files, there should be plenty of examples of new boards being added via PR's.

I usually use platformio to build, so I'm not sure if you'll need to modify anything else for building in Arduino.

vaidyasr, 2018-08-15:
The relay am using is different from the thread I have attached. My relay is YwRobot and whenever I am connecting the esp8266, the switches were default on and not sure why. Ordered a new 4 ch relay from aliexpress and will test it for generic functionality.

ColinShorts, 2018-10-14:
I missed your last reply. Did the new module work? Did you try flipping between INVERSE and NORMAL for the delay provider?
If you can share links to the devices and post the config you use it might help us help you 😀